betaflight/betaflight #6533: BlackBox Doesn't work on FPVM_BETAFLIGHTF7  Verson 3.5 RC2

Strangedave, 2018-08-08:
[BTFL_BLACKBOX_LOG_WIN_5_20180808_164110.zip](https://github.com/betaflight/betaflight/files/2272241/BTFL_BLACKBOX_LOG_WIN_5_20180808_164110.zip)
[Diff 3-5-0 v1.txt](https://github.com/betaflight/betaflight/files/2272245/Diff.3-5-0.v1.txt)

**Describe the bug**
I just flashed my FC with 3.5 RC2 this afternoon. I used the cli to restore everything from my 3.4 configuration. Changed the f_pitch = 110, f_roll = 110, and f_yaw = 110 basically because that was where my d setpoint weight was at before. This machine is flying great. 8 Packs, ran 3 logs, and only 1 will open in either plazmatree or blackbox explorer. the log above did not work. the one below did. The log that worked was with default  f_pitch = 60, f_roll = 60, and f_yaw = 60. I think that was the only change. 
[BTFL_BLACKBOX_LOG_WIN_5_20180808_133155.zip](https://github.com/betaflight/betaflight/files/2272261/BTFL_BLACKBOX_LOG_WIN_5_20180808_133155.zip)
 

**To Reproduce**
record bb log in flight.

**Expected behavior**
log crashes plazmatree and bb explorer

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
Betaflight F7 from race day quads
DALRC Engine esc
Tramp HV VTX
Xhover Win5  Frame
T motor F40Pro II
HQ Prop 5X4.5X3 V1S
Tattu 1300 Rline 100C

etracer65, 2018-08-09:
The log files appear to be corrupted. You're logging at quite a high rate so I would recommend trying with `set blackbox_p_ratio = 64` or maybe `set blackbox_p_ratio = 32`.

Strangedave, 2018-08-09:
I’ll try that tomorrow, it was working in 3.4.0 I never upgraded to 3.4.1

McGiverGim, 2018-08-09:
It seems more of this problem https://github.com/betaflight/betaflight/issues/6490

The header is mixed with P and I frames. In my log it happened the same but it started near the end of the header. Here happens early.

Strangedave, 2018-08-10:
Update - blackbox_p_ratio = 64 ran new log, it also will not open. Plazmatree just shuts down, and Blackbox explorer says "Error opening log: No logs in this file could be parsed successfully". The rig is running great though. This is the Diff from today.
[Diff 3-5-0 v2.txt](https://github.com/betaflight/betaflight/files/2279233/Diff.3-5-0.v2.txt)
And this is the log. 
[BTFL_BLACKBOX_LOG_WIN_5_20180810_135230.zip](https://github.com/betaflight/betaflight/files/2279234/BTFL_BLACKBOX_LOG_WIN_5_20180810_135230.zip)

Strangedave, 2018-08-12:
Update - blackbox_p_ratio = 32 ran new log, this one was ok. log file below.

[BTFL_BLACKBOX_LOG_WIN_5_20180811_212649.zip](https://github.com/betaflight/betaflight/files/2280740/BTFL_BLACKBOX_LOG_WIN_5_20180811_212649.zip)

closed 2018-08-12 by Strangedave

yoodog, 2019-09-01:
similar issue on 4.0.5! odd. will duplicate my comment here

yoodog, 2019-09-01:
same problem with me on 4.0.5 now. Writing to SD card on built in port.
I got a file for each flight 10-23mb - 3-6min flights. But not a single one can be open.
I run f745 board Omnibus v2 f7- 8/8 dshot600 and RPM Filtering + standard 4.x.x filtering

blackbox set to 2khz, SD card = card is ready, and well it records
black box switch is AUX2, same as ARM switch, so records every time I arm
set debug_mode = GYRO_SCALED
blackbox_p_ratio = 64 (I didn't touch it, guess it it is default, but mentioning it as it seems to be a key of some sort)
my diff all incl defaults and dump
[Diff all defaults.txt](https://github.com/betaflight/betaflight/files/3563933/Diff.all.defaults.txt)
[dump full - not all may be good for your controller.txt](https://github.com/betaflight/betaflight/files/3563934/dump.full.-.not.all.may.be.good.for.your.controller.txt)

log file
[LOG00012.BFL.zip](https://github.com/betaflight/betaflight/files/3563932/LOG00012.BFL.zip)

can any one advice

McGiverGim, 2019-09-02:
@yoodog your log file is corrupted. It does not have the Blackbox header.
If you are using a SD Card, try formatting the SD Card again using this tool https://www.sdcard.org/downloads/formatter/ . This sometimes fixed my problem.

yoodog, 2019-09-02:
Thanks man, that’s what I figured as possible issue also last night, so I reformatted with that tool. Will check and report back later today if that fixes it