betaflight/betaflight #6534: Artificial Horizon Reversed

DonaldHillNZ, 2018-08-09:
Hi, Im not sure if this has been covered since last year but I've just discovered that the artificial horizon is working the wrong way around in BF 3.4.1. I have it installed on two different FC's a Matek F405-CTR and a CL Racing F4. Both are showing in Betaflight that the Artificial Horizon in the setup screen is operating the opposite to the attitude of the FC.

DonaldHillNZ, 2018-08-09:
And an addition that might or might not be connected since it's only with the Matek F405-CTR I have the position of the artificial horizon way down at the bottom of the screen. I can adjust this in a way with the set command but something seems wrong.

mikeller, 2018-08-09:
@DonaldHillNZ: That's probably because you positioned it down there - the position is adjustable on the 'OSD' tab in configurator.

DonaldHillNZ, 2018-08-09:
Nope when you try and drag it up it only goes as far as the top most position which is sort of a third way up. Ive managed to move it around a bit using the Set commands but again only to about a third up. Very weird. Oh well it was a nice to have not at all essentual

DonaldHillNZ, 2018-08-09:
essential even

DonaldHillNZ, 2018-08-09:
ever tried to move the ah. does work after lots of trial and error but jumps all over the place. Moving to where to want it them letting the grab go can mean anything. same with the hor bars

jflyper, 2018-08-13:
I think there is a bug in the configurator OSD element GUI layout code for multi-line elements; when the object is grabbed and moved even a single character in any direction, the grab point would become the center of the object. It is also extremely hard or impossible to place the object at extremely high or low positions.

I will try to file an issue in BFC repo.

DonaldHillNZ, 2018-08-14:
Yes I agree. Its very difficult to line it up. At first I thought it 
actually wasnt moving but after lots of perseverance I managed to sort 
of line everything nearly in position.

Let me know if you do file an issue.

Regards

Donald

PS. For some reason some seem to look down on us that want to use 
it..... Oh well takes all sorts.

<mention@noreply.github.com>
Sent: 13/08/2018 9:45:39 PM

closed 2018-09-24 by stale[bot]