betaflight/betaflight #6543: OMNIBUSF4SD Bootloop [4S Battery]

padok, 2018-08-10:
**Describe the bug**
On OMNIBUSF4SD
Bootloop when using 4S (14.8V) Battery's at VBAT port.

**To Reproduce**
Connect Battery directly to VBAT port test if measurement is correct with 3S (11.1V) Battery's.
Now Connect the 4S Battery and after 1-2 sec the FC crashes and starts boot looping.

 **Expected behavior**
Don't crash when using 4S Battery's.

**Flight controller configuration**
clean flash no configuration. (Full chip erase) v3.4.1

**Setup / Versions**
- Betaflight OmniBus F4 V3 Flight Control Built-in OSD/BEC Chip For FPV Drone D
  - [ebay.de;  top_rc001 (8465); WONG KI SHING, 2/F NO780 SHUN TAT STREET, SHUN FUNG WAI ,TUEN MUN, default default, HK]
 - PDB-XPW W/ 5V 12V Dual BEC 140A Current Sensor für FPV Racing Quadcopter RC541
--

--

- 5V PDB -> Omnibus 5V
- VCC PDB -> Omnibus VBAT
- Curr PDB -> Omnibus CNNT 

**Additional context**
I also tried to use two 10k ohm resistors as a voltage divider at Omnibus VBAT but unfortunately the desired effect did not occur.

etracer65, 2018-08-10:
I just tested a bare OmnibusF4 v3 using target OMNIBUSF4SD on both 4S and 3S.  No problems observed.  Note that the flight controller was powered by it's on-board BEC from the VBAT connection.

It sounds like you have some kind of hardware problem and your PDB may not be supplying a clean 5V when powered by 3S.  Also I'm not really sure how you have everything wired.  The OmnibusF4 boards I've seen have an on-board 5V BEC that's powered by the VBAT connection.  So if you've also wired a separate 5V supply to the board that doesn't sound proper.  By connecting both sources of 5V power you are likely causing a problem and one or both of the BEC's might be overloading or overheating causing a shutdown and reboot.

jflyper, 2018-08-10:
@etracer65 
Authentic Omnibus F4 V3 allows supplying external 5V via ESC-center rail. There is a diode triplet that selects highest voltage from three possible sources; USB VBUS, onboard 5V regulator and the external 5V via ESC-center rail.

@padok 
That being said, I suspect your board is a counterfeit, and has one of the diode triplet that handles ESC-center rail installed backwards, which is frequently seen on counterfeit boards.
If this is the case, output from 5V rail or USB VBUS will collide with external 5V source and make the 5V inherently unstable.

Check the polarity of the diodes D1, D2 and D3 to backup the above theory.

[image]
(Note: LDO is switching regulator for F4V3.)

padok, 2018-08-10:
Thanks for your fast responses!

I continued testing a little further and found out that I used a jumper that supplied 5V to RAM (to power my camera), as shown in the diagram below left. (RAM pin will be powered by 5V)

[image]
Source: https://blog.dronetrest.com/omnibus-f4-flight-controller-guide/

After removing the jumper the problem disappeared.
Now I have connected a 5V output of my PDB to the RAM pin and grounded the AGND pin, so far it works.

I still don't know exactly why this behavior occurs only when I want to operate my camera via the voltage regulator of the board (only with a 4S and not a 3S LIPO), but I can definitely recommend not to set this jumper at all.

jflyper, 2018-08-11:
@padok 
> but I can definitely recommend not to set this jumper at all.

I think you are over generalizing an edge case.

This is the first time that I've ever heard the power supply related boot loop problem on this board including authentic, counterfeits and sw-compatible clones.

jflyper, 2018-08-11:
@padok 
Is this your board?
https://www.ebay.de/itm/Betaflight-OmniBus-F4-V3-Flight-Control-Built-in-OSD-BEC-Chip-For-FPV-Drone/263853008027?hash=item3d6edd649b:g:A~cAAOSw52tbYtGo
Looks like it's authentic ...

jflyper, 2018-08-13:
Closed as the thread starter seems to have the problem solved.

closed 2018-08-13 by jflyper