betaflight/betaflight #6544: Firmware will not boot or stuck in bootloader

FonzyMantooth, 2018-08-10:
I have a Flight Test Gremlin and the flight controller (SPRacing F3 EVO) has been working well. I upgraded to the new Betaflight RC2 version 3.5 firmware (we had an April build, not sure which one) and disabled the "No Reboot Sequence" this way I didn't have to short the boot pads.

The upgrade seemed to go fine and I set up the controller, motors, PID , acc loop rate, arm button, etc. I can see the controller meters move when I move the controller sticks and the motors turn when I used the sliders on the motor tab. However, I observed the following behavior

Plugged into the Battery, no Computer:
I plug in the battery, the boot song plays and I get a solid blue light but no red flashing light appears and I can't get the controller to arm. 

No battery, plug into the computer
I plug it into the computer and I get a solid blue light, no red flashing lights. However, the configuration loads and I can see the controller meters work when I move the controller sticks. I can even move the drone and the rendering moves with it in the calibration tab. 

I actually have two of these drones and the first one worked just fine the second one is in this odd state that seems like it is trying to be in a boot mode. 

When I updated the controller on the problem drone I used the "no boot sequence" option. When I upgraded the drone that works, I went to boot mode the old fashioned way and shorted the pin's. That is the only difference in the approach. 

The blue light makes it seem like the system wants to be in boot mode as if a bit is stuck somewhere. 

Finally, I'm able to reflash other firmware (I tried older and newer firmware) and I used the erase flash option several times.   But the same issue is still happening.

Thanks

etracer65, 2018-08-10:
Sounds like you just need to determine the arming disabled reason.

WITH PROPS REMOVED! Connect the flight battery and also connect the USB and connect with the configurator.  Go to the CLI and type `status`.  Note the arming disable flags.  "CLI" and "MSP" are normal, anything else indicates your problem.

etracer65, 2018-08-10:
No, the flight controller firmware has no control over voltages.  It's more likely that electrical noise from the ESC's and motors is at fault.  Some ESC's can cause voltage spikes during active braking.  The solution to this is to add low-ESR capacitors to the power system to absorb thos spikes.  Alternately power your VTX off of filtered power source like a PDB instead of directly from the battery.

FonzyMantooth, 2018-08-11:
Thanks for the help

closed 2018-08-11 by FonzyMantooth