betaflight/betaflight #6548: beeper doesnt beep with rx_lost on

m013, 2018-08-11:
**Describe the bug**
beeper doesnt beep repeatedly when a radio is turned off, only beeps once

**To Reproduce**
enable rx_lost for beeper, disable rx_set

**Expected behavior**
1)i want a beeper and motors (dshot command) to beep with rx_set on (turn it on via switch) simultaneously
2)in case of signal lost only beeper should beep  repeatedly (rx_lost on) 

**Flight controller configuration**
```
diff all

# version
# Betaflight / DYSF4PRO (DYS4) 3.4.0 Jul  8 2018 / 04:07:49 (cce8a59be) MSP API: 1.39

board_name DYSF4PRO
manufacturer_id 
mcu_id 003100363436510536323332
signature 

# reset configuration to default settings
defaults nosave

# name
name Leopard

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09
resource CAMERA_CONTROL 1 A01

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon

# map
map AERT1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1075 0
aux 1 1 1 1375 1625 0
aux 2 4 1 1925 2100 0
aux 3 13 3 1350 1650 0
aux 4 26 3 900 1100 0
aux 5 28 1 1900 2100 0
aux 6 35 1 900 1100 0
aux 7 39 0 1875 2100 0

# adjrange

# rxrange
rxrange 0 987 2010
rxrange 1 987 2010
rxrange 2 987 2010
rxrange 3 988 2011

# vtx

# rxfail
rxfail 5 s 1500
rxfail 7 s 1500

# master
set gyro_lowpass2_hz = 0
set gyro_overflow_detect = OFF
set yaw_spin_threshold = 1200
set acc_trim_pitch = 4
set acc_trim_roll = -14
set acc_calibration = -198,44,-93
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 7
set rc_interp = MANUAL
set rc_interp_int = 17
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set min_throttle = 1010
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 15
set failsafe_off_delay = 30
set failsafe_throttle = 1130
set failsafe_throttle_low_delay = 150
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 32
set vbat_scale = 111
set ibata_scale = 502
set ibata_offset = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set osd_rssi_alarm = 25
set osd_cap_alarm = 1300
set osd_vbat_pos = 2402
set osd_rssi_pos = 2424
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2434
set osd_remaining_time_estimate_pos = 461
set osd_flymode_pos = 34
set osd_throttle_pos = 259
set osd_vtx_channel_pos = 364
set osd_ah_pos = 199
set osd_current_pos = 2048
set osd_mah_drawn_pos = 56
set osd_craft_name_pos = 35
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 425
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 489
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2073
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set vtx_channel = 8
set vtx_freq = 5880

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set crash_delay = 10
set crash_recovery = ON
set smart_feedforward = ON
set iterm_relax = RP
set throttle_boost = 1
set throttle_boost_cutoff = 50
set p_pitch = 67
set i_pitch = 72
set d_pitch = 29
set p_roll = 63
set i_roll = 63
set d_roll = 28
set i_yaw = 100

# profile
profile 1

set dterm_lowpass_hz = 110
set dterm_notch_cutoff = 0
set anti_gravity_gain = 1500
set crash_delay = 10
set crash_recovery = ON
set iterm_rotation = OFF
set smart_feedforward = ON
set iterm_relax = RP
set p_pitch = 65
set i_pitch = 65
set d_pitch = 31
set p_roll = 60
set i_roll = 60
set d_roll = 31
set d_yaw = 8
set abs_control_gain = 10

# profile
profile 2

set anti_gravity_gain = 1500
set p_pitch = 68
set i_pitch = 73
set d_pitch = 36
set p_roll = 60
set i_roll = 65
set d_roll = 36
set p_yaw = 75
set i_yaw = 55
set d_yaw = 8

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 141
set pitch_rc_rate = 141
set yaw_rc_rate = 144
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 0

# rateprofile
rateprofile 1

set thr_mid = 0
set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 143
set roll_srate = 58
set pitch_srate = 58
set yaw_srate = 58
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 2

set thr_mid = 0
set roll_rc_rate = 137
set pitch_rc_rate = 137
set yaw_rc_rate = 137
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
# 

```
**Setup / Versions**
 - dys f4 pro v2
 - r-xsr, taranis x-lite
 - sbus + smartport via softserial

mikeller, 2018-08-11:
Try

```
beeper RX_LOST_LANDING
beacon RX_SET
```

m013, 2018-08-11:
@mikeller , nothing changed

m013, 2018-08-11:
@mikeller , nothing changed.
beeps once again when i turn off the radio# diff all

# version
# Betaflight / DYSF4PRO (DYS4) 3.5.0 Aug  2 2018 / 01:56:30 (d0a9a0fd9) MSP API: 1.40

board_name DYSF4PRO
manufacturer_id 
mcu_id 003100363436510536323332
signature 

# reset configuration to default settings
defaults nosave

# name
name Leopard

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09
resource CAMERA_CONTROL 1 A01

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -RX_LOST
beeper -ARMING_GPS_FIX
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon
beacon RX_SET

# map
map AERT1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1075 0 0
aux 1 1 1 1375 1625 0 0
aux 2 4 1 1925 2100 0 0
aux 3 13 3 1350 1650 0 0
aux 4 26 3 900 1100 0 0
aux 5 28 1 1900 2100 0 0
aux 6 35 1 900 1100 0 0
aux 7 39 0 1875 2100 0 0

# adjrange

# rxrange
rxrange 0 987 2010
rxrange 1 987 2010
rxrange 2 987 2010
rxrange 3 988 2011

# vtx

# rxfail
rxfail 5 s 1500

# master
set gyro_lowpass2_hz = 0
set gyro_overflow_detect = OFF
set yaw_spin_threshold = 1200
set dyn_notch_quality = 50
set acc_trim_pitch = 8
set acc_trim_roll = -20
set acc_calibration = -198,44,-93
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 7
set rc_interp = MANUAL
set rc_interp_int = 17
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set min_throttle = 1010
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 15
set failsafe_off_delay = 30
set failsafe_throttle = 1130
set failsafe_throttle_low_delay = 150
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 32
set vbat_scale = 111
set ibata_scale = 502
set ibata_offset = 4
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set osd_rssi_alarm = 25
set osd_cap_alarm = 1300
set osd_vbat_pos = 2402
set osd_rssi_pos = 2424
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2434
set osd_remaining_time_estimate_pos = 461
set osd_flymode_pos = 34
set osd_throttle_pos = 259
set osd_vtx_channel_pos = 364
set osd_ah_pos = 199
set osd_current_pos = 2048
set osd_mah_drawn_pos = 56
set osd_craft_name_pos = 35
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 425
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 489
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2073
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set vtx_channel = 8
set vtx_freq = 5880

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set crash_delay = 10
set smart_feedforward = ON
set iterm_relax = RP
set throttle_boost = 1
set p_pitch = 67
set i_pitch = 72
set d_pitch = 29
set f_pitch = 100
set p_roll = 63
set i_roll = 63
set d_roll = 28
set f_roll = 100
set i_yaw = 100
set f_yaw = 100

# profile
profile 1

set dterm_lowpass_hz = 110
set dterm_notch_cutoff = 0
set anti_gravity_gain = 1500
set crash_delay = 10
set crash_recovery = ON
set iterm_rotation = OFF
set smart_feedforward = ON
set iterm_relax = RP
set p_pitch = 65
set i_pitch = 65
set d_pitch = 31
set p_roll = 60
set i_roll = 60
set d_roll = 31
set d_yaw = 8
set abs_control_gain = 10

# profile
profile 2

set anti_gravity_gain = 1500
set p_pitch = 68
set i_pitch = 73
set d_pitch = 36
set p_roll = 60
set i_roll = 65
set d_roll = 36
set p_yaw = 75
set i_yaw = 55
set d_yaw = 8

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 141
set pitch_rc_rate = 141
set yaw_rc_rate = 144
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 0

# rateprofile
rateprofile 1

set thr_mid = 0
set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 143
set roll_srate = 58
set pitch_srate = 58
set yaw_srate = 58
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 2

set thr_mid = 0
set roll_rc_rate = 137
set pitch_rc_rate = 137
set yaw_rc_rate = 137
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

mikeller, 2018-08-11:
Now you have changed your configuration in addition to what you were told to do:

`beeper -RX_LOST`

I really can't help you if you do not follow instructions.

m013, 2018-08-11:
okay. now only motors beep

`# diff all

# version
# Betaflight / DYSF4PRO (DYS4) 3.5.0 Aug  2 2018 / 01:56:30 (d0a9a0fd9) MSP API: 1.40

board_name DYSF4PRO
manufacturer_id 
mcu_id 003100363436510536323332
signature 

# reset configuration to default settings
defaults nosave

# name
name Leopard

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09
resource CAMERA_CONTROL 1 A01

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -ANTI_GRAVITY
feature RX_SERIAL
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -ARMING_GPS_FIX
beeper -RX_SET
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon
beacon RX_SET

# map
map AERT1234

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200
serial 5 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color
mode_color 6 1 2

# aux
aux 0 0 0 900 1075 0 0
aux 1 1 1 1375 1625 0 0
aux 2 4 1 1925 2100 0 0
aux 3 13 3 1350 1650 0 0
aux 4 26 3 900 1100 0 0
aux 5 28 1 1900 2100 0 0
aux 6 35 1 900 1100 0 0
aux 7 39 0 1875 2100 0 0

# adjrange

# rxrange
rxrange 0 987 2010
rxrange 1 987 2010
rxrange 2 987 2010
rxrange 3 988 2011

# vtx

# rxfail
rxfail 5 s 1500
rxfail 7 s 1500

# master
set gyro_lowpass2_hz = 0
set gyro_overflow_detect = OFF
set yaw_spin_threshold = 1200
set dyn_notch_quality = 50
set acc_trim_pitch = 8
set acc_trim_roll = -20
set acc_calibration = -198,44,-93
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 7
set rc_interp = MANUAL
set rc_interp_int = 17
set rc_smoothing_type = FILTER
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set min_throttle = 1010
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT600
set motor_pwm_rate = 480
set failsafe_delay = 15
set failsafe_off_delay = 30
set failsafe_throttle = 1130
set failsafe_throttle_low_delay = 150
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 32
set vbat_scale = 111
set ibata_scale = 502
set ibata_offset = 4
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 2
set osd_rssi_alarm = 25
set osd_cap_alarm = 1300
set osd_vbat_pos = 2402
set osd_rssi_pos = 2424
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2434
set osd_remaining_time_estimate_pos = 461
set osd_flymode_pos = 34
set osd_throttle_pos = 259
set osd_vtx_channel_pos = 364
set osd_ah_pos = 199
set osd_current_pos = 2048
set osd_mah_drawn_pos = 56
set osd_craft_name_pos = 35
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 425
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 489
set osd_disarmed_pos = 2411
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 2073
set osd_esc_rpm_pos = 83
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_bb_no = OFF
set debug_mode = GYRO_SCALED
set vtx_channel = 8
set vtx_freq = 5880

# profile
profile 0

set dterm_notch_cutoff = 0
set anti_gravity_gain = 2000
set crash_delay = 10
set smart_feedforward = ON
set iterm_relax = RP
set throttle_boost = 1
set p_pitch = 67
set i_pitch = 72
set d_pitch = 29
set f_pitch = 100
set p_roll = 63
set i_roll = 63
set d_roll = 28
set f_roll = 100
set i_yaw = 100
set f_yaw = 100

# profile
profile 1

set dterm_lowpass_hz = 110
set dterm_notch_cutoff = 0
set anti_gravity_gain = 1500
set crash_delay = 10
set crash_recovery = ON
set iterm_rotation = OFF
set smart_feedforward = ON
set iterm_relax = RP
set p_pitch = 65
set i_pitch = 65
set d_pitch = 31
set p_roll = 60
set i_roll = 60
set d_roll = 31
set d_yaw = 8
set abs_control_gain = 10

# profile
profile 2

set anti_gravity_gain = 1500
set p_pitch = 68
set i_pitch = 73
set d_pitch = 36
set p_roll = 60
set i_roll = 65
set d_roll = 36
set p_yaw = 75
set i_yaw = 55
set d_yaw = 8

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 141
set pitch_rc_rate = 141
set yaw_rc_rate = 144
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 0

# rateprofile
rateprofile 1

set thr_mid = 0
set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 143
set roll_srate = 58
set pitch_srate = 58
set yaw_srate = 58
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 2

set thr_mid = 0
set roll_rc_rate = 137
set pitch_rc_rate = 137
set yaw_rc_rate = 137
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 67
set tpa_rate = 0
set tpa_breakpoint = 1050

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save`

m013, 2018-08-11:
@mikeller  , i found what the problem was.
1)if i power up a quad with the radio on, turn off the radio - a quad beeps once
2)if i arm a quad, turn off the radio - a quad start to beep after failsafe procedure normally.

so i guess it should be armed once in order it to work correctly

closed 2018-08-11 by m013