betaflight/betaflight #6551: CPU load 100% OmniNXT 32K 3.5 RC2

mooiweertje, 2018-08-11:
I upgraded to 3.5RC2 on my OmniNXT and my working 3.4.1 config pushes the CPU load to 100%.

Install 3.5RC2 on OnmiNXT and load below config.

I expeced somewhat similar CPU load as with 3.4.1.

```
# version
# Betaflight / OMNINXT7 (ONX7) 3.5.0 Aug  2 2018 / 02:15:40 (d0a9a0fd9) MSP API: 1.40

board_name OMNINXT7
manufacturer_id 

# name
name Real1S

# resources
resource MOTOR 1 B00
resource MOTOR 2 B05
resource MOTOR 3 B01
resource MOTOR 4 B04

# mixer

# servo

# servo mix

# feature
feature ESC_SENSOR

# beeper

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 0 2048 115200 57600 0 115200
serial 1 64 115200 57600 0 115200

# led
led 0 8,9::A:11

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 5
mode_color 7 0 2

# aux
aux 0 0 0 1300 2100 0 0
aux 1 0 1 1300 1700 0 0
aux 2 1 0 1300 1700 0 0
aux 3 13 1 1700 2100 0 0
aux 4 26 3 1700 2100 0 0
aux 5 28 0 1700 2100 0 0
aux 6 35 1 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_calib_duration = 250
set gyro_use_32khz = ON
set gyro_to_use = SECOND
set acc_calibration = 0,0,-2048
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_yaw = -90
set current_meter = ESC
set battery_meter = ESC
set ibata_scale = 600
set beeper_dshot_beacon_tone = 5
set yaw_motors_reversed = ON
set small_angle = 180
set gyro_cal_on_first_arm = ON
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_core_temp_alarm = 90
set osd_vbat_pos = 2498
set osd_rssi_pos = 2274
set osd_tim_2_pos = 2518
set osd_flymode_pos = 233
set osd_vtx_channel_pos = 2082
set osd_current_pos = 2088
set osd_mah_drawn_pos = 2097
set osd_altitude_pos = 244
set osd_pid_yaw_pos = 233
set osd_warnings_pos = 2505
set osd_disarmed_pos = 2283
set osd_esc_tmp_pos = 225
set osd_esc_rpm_pos = 2295
set osd_core_temp_pos = 2104
set osd_stat_max_spd = OFF
set osd_stat_min_rssi = OFF
set osd_stat_bb_no = OFF
set vtx_band = 2
set vtx_power = 3
set vtx_freq = 5733
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 66
set i_pitch = 52
set d_pitch = 28
set p_roll = 54
set i_roll = 46
set d_roll = 20
set p_yaw = 70
set i_yaw = 120
set d_yaw = 5

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 30
set roll_rc_rate = 200
set pitch_rc_rate = 200
set yaw_rc_rate = 200
```

DieHertz, 2018-08-11:
Not a bug.
Features of 3.5 increase CPU have intensity

mooiweertje, 2018-08-11:
I'm sorry to hear that. Got to live with that then.
Thanks

mooiweertje, 2018-08-11:
If I can help in any way plz let me know.

Dropsleutel, 2018-08-12:
I had also few times 100% cpu load on F4 3.5Rc2, I found out it was connectedto ledstrip, disabling ledstrip load went back to 10%, you can try?

mikeller, 2018-08-13:
@Dropsleutel: I tested with 3.5.0-RC2, and was not able to replicate that enabling LED_STRIP caused noticeable increase in load.

Can you please provide a config `diff` for the configuration that causes the load to spike for you when LED_STRIP is enabled?

mooiweertje, 2018-08-13:
Turning of the accelerometer drops the CPU load to 91% ±1. Tuning of the led light saves another 1%.
I tried yesterday and it flies stable
32/32 kHz at 90±1% CPU load.

2.5 flies much better than 3.4.n. I was having problems with yaw drifting which is almost if not completely gone now.

mikeller, 2018-08-20:
@controlflow: The flight performance improvements in 3.5 have added more complexity to the gyro / PID loop calculations, hence there is an increase in load - you can't get something from nothing as they say...

mooiweertje, 2018-08-21:
When I change the PID loop frequency from 16KHz to 32 Khz the CPU load goes
from 20% to 90%. Changing both PID and Gyro loop from 16KHz  to 32Khz makes
the load go from 8% to 90%.

I haven't seen the code but this seems out of proportions?

mikeller, 2018-08-21:
@mooiweertje: The 'load' that is calculated in Betaflight is indicative of the actual load (at best), and a far way off from being linear.

mooiweertje, 2018-08-21:
It seems to jump from 90% to max very rarely. I can try to optimize it a bit for my FC or is that a lost cause?

DieHertz, 2018-08-21:
What makes you think load scales linearly with sampling frequencies?

closed 2018-09-27 by stale[bot]