betaflight/betaflight #6553: MATEK F722 32K Pid loop 86% CPU load

simoccc, 2018-08-12:
I can confirm that even on Matek F722 with DSHOT1200 accelerometer, pid 32k and gyro 32k the cpu load is very high ( 86%) .

If I lower the pid loop to 16k the cpu load goes to 26% , that is a real big difference , is it normal ? 

With 3.4.1 same settings I was around 32%

flint723, 2018-08-13:
See #6551.  If 32k32k is essential the only solution is go back to 3.4.1 (for now).

fujin, 2018-08-13:
You can also overclock to 240mHz for 32/32/32 (dshot1200), but otherwise for now, we have lost a little bit of the headroom that had been previously established.

etracer65, 2018-08-13:
Try disabling the BARO.  The barometer on that board uses I2C to connect and it adds a lot of load to the processor.

etracer65, 2018-08-13:
I just tested that board and using 3.5RC2, DSHOT1200, ACC off, BARO off and overclocked to 240MHz it runs at 30% with 32/32K.  Without overclocking it's about 50%.

simoccc, 2018-08-13:
Hi, I have already barometer disabled , here is my diff , with this configuration I have 26% of CPU , if I switch PID loop to 32K ( instead of 16K ) it jumps up to 86%:

```
# Betaflight / MATEKF722 (MKF7) 3.5.0 Aug  2 2018 / 02:10:13 (d0a9a0fd9) MSP API: 1.40

board_name MATEKF722
manufacturer_id 

# name
name CHIMERA

# resources
resource MOTOR 6 NONE
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature LED_STRIP
feature AIRMODE
feature ESC_SENSOR

# beeper
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 2 32 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1925 2100 0 0
aux 1 1 2 1900 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 35 3 1825 2100 0 0
aux 4 39 4 1825 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_use_32khz = ON
set mag_hardware = NONE
set baro_hardware = NONE
set motor_pwm_protocol = DSHOT1200
set pid_process_denom = 2
set osd_vbat_pos = 2092
set osd_rssi_pos = 2106
set osd_flymode_pos = 2145
set osd_vtx_channel_pos = 2520
set osd_current_pos = 2465
set osd_craft_name_pos = 2506
set osd_avg_cell_voltage_pos = 2081
set osd_esc_tmp_pos = 2209
set osd_esc_rpm_pos = 2177
set vtx_band = 3
set vtx_channel = 3
set vtx_freq = 5665

# profile
profile 0

# rateprofile
rateprofile 0
```

etracer65, 2018-08-13:
Nothing abnormal about that.  You can free up some processing if you disable the accelerometer.  Also it looks like you have LEDSTRIP enabled but haven't defined any LED's.  If you want to reduce the cpu load you can try using the `cpu_overclock` option but consider that possibly unstable and you'll have to test for yourself.

simoccc, 2018-08-13:
in 32K/32K mode ( PID / Gyro ) without Accelerometer CPU runs at 67% instead of 86% that is something.

LedStrip and ESC Sensor weight is about 3% each .
I prefer not to overclock in this moment!

I'll try to fly and see what I prefer between PID loop 16K on 3.5 or 32/32 on 3.4.1 .

I open this issue because I wanted to give you some feedback and I hope you can optimize the code for F7 processor further to give all the options for it.

Let me know if you need some comparison or test, or I if you prefer I can close the issue

Best Regards

DieHertz, 2018-08-13:
It's optimized already and is capable of more than F4

ykyandy, 2018-08-13:
With 3.4.1 same settings but  I was around +_91%

Diaoul, 2018-08-26:
I have this issue as well. Is that OK to run 32/32 at 90% CPU? What's CPU load is considered "safe"?

DieHertz, 2018-08-26:
No, it's not.

closed 2018-10-02 by stale[bot]