betaflight/betaflight #6554: Crazy Landing - DYS F4 - Betaflight 3.5.0 RC2

xudaondo, 2018-08-13:
**Describe the bug**
When landing, with and without airmode, when the quad get contact with the floor (very slowly) the quad gets crazy, full throttle and a lot of flips in the floor.

Then, the motors stops, and it doesn't arm again.

**To Reproduce**
FC: DYS F4
Betaflight Version: 3.5.0 RC2 Target DYSF4

**Expected behavior**
Correct landing, like in Betaflight 3.4.1 with target AIRBOT F4

**Flight controller configuration**
Pending

jflyper, 2018-08-13:
@xudaondo 

Sounds like an isolated config problem case.

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

closed 2018-08-13 by jflyper

xudaondo, 2018-08-13:
But it is very strange, that with Betaflight 3.4.1 I don't have this problem.
It only happens with the new 3.5 RC2.

All the settings by default, except de specific ones for dshot and my receiver....