betaflight/betaflight #6556: TPA curves like RF/RF1/Flight1

kenkamm, 2018-08-13:
Betaflight does not give the user full control of the shape of the TPA curve. P, I, and D attenuation should be independently configurable and a defined by a multipoint curve. The current system doesn’t not allow a bathtub curve which would be beneficial for many quads. It would be especially useful in battling MTO (mid throttle oscillations).

The Butterflight team has released a initial version of TPA curves for the HelioRC Spring FC.

DaveFPV, 2018-08-13:
Yep, that would be nice.

theSkonk, 2018-08-15:
im a big fan of TPA and more control of it... but with BF 3.5 its less nessasary.. I'm finding moving from 4s to 6s or different props ( where TPA was useful ) there is no need to tune the quad any different.. i do recommend setting the TPA breakpoint down to 1050 though as it helps with MTO if your having them

kenkamm, 2018-08-15:
On 5” machines that are imperfect it can really help to get the jello out of the GoPro. It’s very, very useful on 6” or 7” machines where it’s much more challenging to get rid of mid throttle oscillations.

kenkamm, 2018-08-19:
Code for this has been implemented in Butterflight for the HelioRC Spring (maybe all BuF but I’m not sure), written by apoc. Would love to have this for Betaflight.

Ralfde, 2019-03-12:
After testing the BF 4.0.0 RC1, which is really great, this TPA curves, mainly for P is the most missing feature in BF 4.0.0. Maybe this could be added for the official BF 4 release.

theSkonk, 2019-03-12:
@Ralfde RC mean feature freeze.. nothing more added till 4.1  i think you guys aren't getting how much this isn't needed..   D causes pretty much all the issues that people think TPA curves will fix if your flying 4rc1 just try less P.. also you going to get no where around here referencing Apoc or butterflight (RIP)

Ralfde, 2019-03-12:
I'm using less P, but to reduce Propwash the best possible way at all throttle positions TPA curves for P and D will be needed. Please make a pull request for us for BF 4.1.

theSkonk, 2019-03-13:
p doesn't effect prop wash and we are already mainly using D only TPA.. according to some very good pilots changing the PID control settings in F1 has next to no effect any way. i don't see any one who could code this doing it any time soon as its a reasonably pointless feature

Ralfde, 2019-03-13:
I think it's always a combination of P and D. 
I also think the main difference would be at very low throttle positions, mainly min throttle and at full throttle. Between the effect is very low and can be done with the existing methods.

theSkonk, 2019-03-13:
we have spent a lot of time working on trying to best address prop wash.. working with tuning of the D_min, the D only TPA, filter tuning and there is no reason to go back to TPA that effects I  or P.  things to make prop wash go away... are low filtering latency, high powered motors with light props  ( improve response time ), using enough D, using enough D_min gain, actually flying the quad properly ( this is what the so called 'pro' FPV pilots do they avoid zero throttle situations and fly around the things that are known prop wash triggers and recently Rc filtering settings. P and P based TPA is not something thats come to light across literally thousands of hours of flight testing and BB logging

Ralfde, 2019-03-13:
Thanks a lot. After reading you comment I did check the Betaflight Wiki about these parameters.
This will help me a lot and I do now understand a lot more about the Betaflight PIDs.
d_min_gain is not available any more. Is this replaced by d_min_boost_gain?

butterSnake, 2019-03-13:
Yeah I think that’s the new term. I’ve been using a gain of 30. This brings in the extra D Nice when prop wash starts

Ralfde, 2019-03-14:
Thanks a lot. I think, betaflight could also measure the motor torque by the new possibility of rpm telemetry of the bidirectional dshot and could make an automatic tpa later on, if needed. Lots of possibilities with the bidirectional dshot.

Ralfde, 2019-03-17:
Your help about the d_min was a great information. No more prop wash. Thanks a lot.

closed 2026-09-26 by blckmn