betaflight/betaflight #6557: AUX Switch beeper not working

Deckburner, 2018-08-13:
**Describe the bug**
A clear and concise description of what the bug is.
BF 3.4.1
Target: CLRacingF4 

On POWER & BATTERY tab of Betaflight the VOLTAGE METER SOURCE was set to NONE.
When I set up an AUX switch for a beeper/buzzer, the beeper would not sound.
**a lipo batter was plugged into the quad when the AUX switch-beeper was attempted**

When I changed the VOLTAGE METER SOURCE to ONBOARD ADC, the AUX Switch, when set in the "armed" position now sounds the beeper.
**again, a lipo batter was plugged into the quad when the AUX switch-beeper was attempted**

**To Reproduce**
Steps to reproduce the behavior:
In BF:
1) Set up an AUX Switch to a beeper.
2) POWER & BATTERY tab
--Set VOLTAGE METER SOURCE to NONE
--Switch AUX switch (for a beeper) to sound the beeper
--Beeper should NOT sound
**a lipo batter must be connected to the quad for the beeper to sound.  The beeper will not sound with only the micro-USB plugged in**

**Expected behavior**
A clear and concise description of what you expected to happen.
I did not think that the VOLTAGE METER SOURCE should affect the beeper and the AUX switch for it.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
```
diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.4.1 Aug  2 2018 / 07:49:06 (1d99444d9) MSP API: 1.39

board_name CLRACINGF4
manufacturer_id 

# name
name Deckburner - Cha

# resources

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature RSSI_ADC
feature AIRMODE

# beeper
beeper -ARMED
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 3 2048 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0
aux 1 13 2 900 1300 0

# adjrange
adjrange 0 0 1 900 2100 12 1 0 0

# rxrange

# vtx

# rxfail

# master
set min_check = 1010
set max_check = 1990
set rssi_invert = ON
set serialrx_provider = SPEK2048
set motor_pwm_protocol = DSHOT600
set ibata_scale = 243
set beeper_frequency = 0
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_rssi_alarm = 10
set osd_vbat_pos = 2433
set osd_rssi_pos = 2081
set osd_tim_2_pos = 2455
set osd_flymode_pos = 364
set osd_mah_drawn_pos = 2401
set osd_warnings_pos = 2440
set osd_disarmed_pos = 2411
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_max_curr = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 4
set vtx_power = 2
set vtx_freq = 5769

# profile
profile 0

set dterm_notch_cutoff = 0
set p_pitch = 47
set i_pitch = 35
set d_pitch = 18
set i_roll = 35
set d_roll = 18
set p_yaw = 63
set i_yaw = 39

# rateprofile
rateprofile 1

set roll_expo = 25
set pitch_expo = 25
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 85

# 
```

**Setup / Versions**
 - Flight controller  [what type is it, where was it bought from];
CLRacingF4S (V1.4) bought from ArmattanQuads
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
RX: Lemon DIversity Satellite Receiver LM0037
VTX: TBS Unify Pro HV
 - how are the different components wired up.
all came prebuilt from Armattan.  I believe that all hardware wires directly into the FC.

**Additional context**
Add any other context about the problem here.
This is my first time ever reporting an issue.  I apologize in advance if I did not do an excellent job.  Thank you.

jflyper, 2018-08-13:
@Deckburner 

It is the expected behavior.

As you switch battery meter source between none and onboard ADC, you are turning the meter between off and on, which triggers `-ON_USB` on and off. This is because the `-ON_USB` actually is not a real "_connected with usb_", but it silences the beeper when the FC is not detecting battery voltage.

closed 2018-08-13 by jflyper

Deckburner, 2018-08-14:
OH, dang...so sorry to take up your time on this then.  I didn't mean to
waste anyones time.  Thank you for the explanation.  I appreciate the
insight into how it works.  Thank you for doing what you do so guys like me
have a better experience flying!!
have a great week,
Steve

jflyper, 2018-08-14:
@Deckburner No problem :)