betaflight/betaflight #6558: Reset Rescue mode

CcLaFleche, 2018-08-13:
Hi !

Congratulation for your job, everyone !

Could you add a reset position for rescue mode, for a more safety position of return ?

SchonRichtig, 2018-08-14:
Hi, 

thought about it, but until landing is noch realy implement it is not realy usefull.
In rescue Mode you should take the controll back when you can and land it by yourself,
at least before it hammers into the ground :)

CcLaFleche, 2018-08-14:
Is there a chance to return to the functions of baro and gps?
Because Cleanflight has also removed and Inav is really behind ...!
Every time I go back under inav I have the impression to return 2 years back ...

SchonRichtig, 2018-08-14:
GPS rescue uses the GPS and the Baro if build in to determine the height, if you mean that?
Or do you mean more / other GPS Funktions?

CcLaFleche, 2018-08-14:
Baro end gps hold positioning, useful for long range

CcLaFleche, 2018-08-14:
A long range quad is not a hight power quad, it's an efficient quad.
What Betaflight tries to achieve finaly, ? If you can talk about this...

mikeller, 2018-08-14:
@CcLaFleche: Betaflight is focused on high powered quads for racing and acrobatics.

CcLaFleche, 2018-08-14:
Betaflight is very innovative, compatible with the largest number of controllers available on the market and is therefore intended for a wide audience. It will be a pity that he locks himself into a single lane while the talented people behind him allow so much functionality
But that's just my point of view.

Thanks for your answer and your time.

mikeller, 2018-08-14:
@CcLaFleche: It's not so much a matter of 'locking ourselves in', it is more a matter of 'choosing our battles' - everybody behind an open source project like Betaflight is doing this in their spare time.
Instead of trying to do everything, and ending up doing everything in a lousy way (like it was the case with the now removed GPS hold and altitude hold functionality), choosing what to do, and doing it well is key, considering that the time we have to do it is limited. This is a key element that you will see repeated in pretty much every successful project.
Also, Betaflight is collaborating closely with iNav, and exchanging a lot of code with this project. This makes us even less inclined to incorporate functionality that iNav already does, and does well.

SchonRichtig, 2018-08-14:
Maybe yout should trie INAV 2.0.0 RC6 or newer.
Everythink i need is that the quad come back into range if i loose rc signal, and that is exactly what gps rescue does.
An "egg-laying wollmilchsau" :) would be nice, but for long(er) range flying i have everything what i need

CcLaFleche, 2018-08-14:
i m already testing the last inav rev... thanks

closed 2018-08-14 by CcLaFleche