betaflight/betaflight #6563: Smart Audio Clicking Sound - not going away after arming BF3.5RC2 OMNIBUS4SD AKK FX2

jasc76, 2018-08-14:
Hi

I hope I do it right, to wite this here like this

Well using BF3.5RC2 OMNIBUS4SD Target 
AKK FX2 VTX 

and I constantly hear the smartaudio click sound even after arming :-(
From what I found that should disappear shortly after arming

Thanks
Jan 

PS: Love Betaflight and all the work that is put into it

jflyper, 2018-08-15:
@jasc76 

Two basic questions.

- Are you using AKK fu**ed up firmware?

- Is the VTX online?

jasc76, 2018-08-15:
I havent messed with the firmware of the AKK. SA is working fine
VTX works also.
I read that there was an SA issue, but since it was talked about it in BF3.3, I thought in 3.5 its still fixed.

jflyper, 2018-08-15:
Hmm... here's what I know.

1. The SA polling never happens when armed. A portion of the program that does the polling is not even called when the controller is armed.
2. The SA polling, as of 3.3, only sustain for a few (one, most of the case) second after initialization and whenever changes are made.
3. Your symptom does not develop with 3.5.0 OMNIBUSF4SD target firmware, with Unify Pro 5G8.

I think you have to hookup the Logic or something on to the SA wire and find out who is talking what.

jasc76, 2018-08-15:
well, I'm not sure. That the noise is SA was suggested by more experienced flyers. 
I added a little sample... Its the end of a flight and noise is there while flying and after disarm
[audio_noise.zip](https://github.com/betaflight/betaflight/files/2289223/audio_noise.zip)

jflyper, 2018-08-17:
@jasc76 

Your noise is not SmartAudio origin.

Clicking noise that is originated from SmartAudio is much slower interval, as the driver software talks to the device only at 5Hz (200msec interval). Your noise is 25Hz (40msec interval).

To verify, disconnect the SmartAudio wire and listen.

closed 2018-09-23 by stale[bot]

jasc76, 2018-09-25:
Sorry, my bad. 
This noise is from Aomway Commander V1 DVR
It seems this is common :-(
Sorry for posting this here, nobody I asked upfront was able to answer this