betaflight/betaflight #6564: MSC not working when blackbox flash is full

ghostface, 2018-08-14:
I can reliably reproduce this. 
When the blackbox log storage is full the mass storage device emulation seems to be broken.
It works reliably every time when it is not full tho.
FC will reboot but mass storage device does not show up.

I am using an Omnibus f4 and bf 3.5rc2 with git configurator and I am using archlinux as OS - will upgrade to final 3.5 asap tho.

jflyper, 2018-08-15:
There are some known problems with cluster and volume size computation in the current FAT32 emulation code for MSC.

ghostface, 2018-08-15:
Thx for fixing the title/label!

Looking forward to a fix - btw. is it possible to delete the blackbox log when using msc, if not any plans/possibility of adding that?

jflyper, 2018-08-15:
@ghostface 
There is no way to delete a file from a data flash device or format the entire device while mounted as a msc FAT32 volume. It requires a complete read/write-able file system on the data flash, which is not trivial at all to implement.

ghostface, 2018-08-15:
@jflyper thanks a bunch, thats what I was assuming!

mikeller, 2018-08-15:
@jflyper: As a suggestion, we could put an emulated `README.TXT` file into the drive explaining these things - we're already using flash space to set a custom icon, so we might as well add some useful information. ;-)

nyanloutre, 2019-05-30:
Same problem for me with Betaflight 3.5.1
Is it fixed in 4.X ?

mikeller, 2019-05-30:
@ghostface: Can you confirm that this has been fixed?

haslinghuis, 2024-05-12:
Stale

closed 2024-05-12 by haslinghuis