betaflight/betaflight #6565: Betaflight 3.5 Lua not working on Fport and CLRacingF4

Footr, 2018-08-15:
**Describe the bug**
When I go to use lua scripts on Betaflight 3.5 they will not load values.  I am running Fport on an R-XSR on a CLRacing F4S FC. 

**To Reproduce**
Steps to reproduce the behavior:  Try to use a lua script, it won't load values.

**Expected behavior**
I expect to see values like I did on Betaflight 3.3

**Flight controller configuration**
```
# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.0 Aug 14 2018 / 03:26:59 (1d710afd7) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 
mcu_id 003000403037471132363731
signature 

# reset configuration to default settings
defaults nosave

# name
name Footr3

# resources
resource MOTOR 5 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 B08
resource LED 1 NONE

# mixer

# servo

# servo mix

# feature
feature MOTOR_STOP
feature TELEMETRY

# beeper
beeper -RX_LOST_LANDING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 2 64 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1800 2100 0 0
aux 2 28 3 1775 2100 0 0
aux 3 35 3 1400 1650 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_hz = 120
set dyn_notch_quality = 15
set dyn_notch_width_percent = 30
set min_check = 1020
set max_check = 1990
set rc_smoothing_type = FILTER
set serialrx_provider = FPORT
set serialrx_inverted = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1050
set vbat_max_cell_voltage = 45
set vbat_warning_cell_voltage = 32
set ibata_scale = 254
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 1200
set osd_tim1 = 529
set osd_vbat_pos = 44
set osd_rssi_pos = 40
set osd_tim_1_pos = 2442
set osd_tim_2_pos = 33
set osd_flymode_pos = 333
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2424
set osd_craft_name_pos = 2401
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 9
set osd_avg_cell_voltage_pos = 2455
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set vtx_band = 5
set vtx_channel = 8
set vtx_freq = 5917
set camera_control_internal_resistance = 480

# profile
profile 0

set dterm_lowpass_hz = 80
set dterm_lowpass2_hz = 160
set dterm_notch_cutoff = 0
set crash_gthreshold = 550
set crash_time = 250
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 51
set i_pitch = 62
set d_pitch = 22
set f_pitch = 100
set p_roll = 50
set i_roll = 58
set d_roll = 20
set f_roll = 110
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 103
set pitch_rc_rate = 90
set yaw_rc_rate = 65
set pitch_expo = 2
set roll_srate = 58
set pitch_srate = 38
set yaw_srate = 43
set tpa_breakpoint = 1550

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; CL Racing F4S
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable]; R-XSR, Tramp, X9D Plus on openTX 2.2 
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

Footr, 2018-08-15:
Yes, I am getting telemetry though I am getting "telemetry lost" messages about every 30 seconds or so.  Fport is connected to the telemetry uart 3.  Yes, nothing has changed hardware wise since I moved from 3.3.

jflyper, 2018-08-15:
Hmm... What if you do "Delete all sensors" and then "Discover new sensors"?

mikeller, 2018-08-15:
@Footr: What firmware version are you using on the R-XSR? If you aren't already using it, try the 180615 'beta' version.

Footr, 2018-08-15:
I'll give the beta firmware a shot, thanks!  Didn't even know they had an update out.

trojanGoat, 2018-08-16:
Thanks.  I had the same problem.  Firmware update seems to fix the issue.  I had to use the 180615 version.  https://www.frsky-rc.com/wp-content/uploads/Downloads/Firmware/R-XSR/Beta/R-XSR_FPORT_180615.zip