betaflight/betaflight #6570: "gyro/pidloop disable" Mode

rephlex, 2018-08-15:
On KISS it is possible to have the quadcopter spin the props at a fixed, low speed with the gyro / pid loop being inactive, as can be seen here starting at 5:28
https://youtu.be/2gwaOW1Pj7c?t=5m28s

This would be helpful IMHO to check for motor problems / noises on the field.

So the feature is: Implement a Mode that when it is active disables the gyro and PID loop.

rephlex, 2018-08-15:
i implemented this and am about to check the changes, will then issue a pull request for you to check out.

AirBreak69, 2018-08-15:
If you arm but do not touch the sticks you could simulate a static motor spin at minimumthrottle by
pid_at_min_throttle = OFF

rephlex, 2018-08-15:
But that's not a Mode, it's static configuration.

Best way IMHO is to add this as a mode, so it won't interfere with the rest of your configuration.

I have a PR ready in for next few hours, a video showing the feature is on its way to YouTube.

rephlex, 2018-08-15:
PR is opened. Video coming in a few minutes.

rephlex, 2018-08-15:
all the explations: https://www.youtube.com/watch?v=iB8sIijXwK4

wind0r, 2018-08-22:
pid_at_min_throttle = OFF + Airmode on a switch is exactlily the behaivor that mr.steele shows with his kiss setup

rephlex, 2018-08-23:
@wind0r although that is correct, the intention is to enable the feature without having to set pid_at_min_throttle, as this will have an influence to normal flying, even with airmode!
when you have pid_at_min_throttle set to OFF, flying with low throttle will disable the PID-controller even in airmode, where it is NOT intended.
Thus this mode which will enable the behaviour without messing with the rest of your config.

rephlex, 2018-08-23:
> Flying in low throttle with airmode and with at min throttle disabled will still have the pid controller active
This is still wrong. Go ahead and test it.

wind0r, 2018-08-23:
https://github.com/betaflight/betaflight/blob/master/src/main/fc/fc_core.c#L595

atomiclama, 2018-08-24:
@rephlex With the test procedure you list above airmode is not being enabled. You need to get the throttle above a set value before airmode is enabled. So do you test again but this time blip the throttle. then go back to min throttle and check that PID loop is on. 

This caught me out in the past when I flew with low throttle from take off and wondered why it lost control when I dropped to min throttle. I fly with pid_min throttle off and airmode on. It works as @wind0r states.

Yamaford, 2018-08-29:
Been running airmode on a switch and cli pid at min off sense airmode was implemented. Works just like steels setup until you get above airmode threshold then it's normal flying until you disarm or turn airmode off.

rephlex, 2018-09-09:
Alright, using the procedure @atomiclama is using is very dangerous. You just neet to raise the throttle a bit to activate airmode/pidloop and chop your fingers.
Hence, another reason why spin at idle mode should be implemented.

atomiclama, 2018-09-10:
@rephlex Sorry you don't get how it works. You can do what is being done in that vid as things are now. You don't need to blip the throttle whilst the quad is in your hand, yes that would be silly.

Turn pid at min throttle off.
Arm and there is no pid active, quad doesn't flip out.
Raise throttle above min_throttle value. You would need todo this to get the quad to hover. PID loop is active but not yet airmode. The 2 things are different. 
Go fly.
Once your throttle is above airmode enable value, airmode is active and you can drop below min throttle and still have pid active so you can do your min throttle flippy floppies.

This is how I fly, I hand launch when the grass is too long or wet or dusty.

rephlex, 2018-09-10:
@atomiclama if you raise the throttle on accident, that's what's dangerous.

atomiclama, 2018-09-10:
If you are doing anything like this then there is the danger of touching the throttle. You do it at your own risk.

rephlex, 2018-09-10:
@atomiclama nevertheless i think minimizing risks does make sense.

That aside, creating thr behaviour your described requires tweaking two command line parameters. Implementing it as a mode is way easier and safer to set up for the end user.

atomiclama, 2018-09-10:
I think we understand each other now so I'm out. 

Leave it to the devs to decide if it is worth while or not.

Cheers.

wind0r, 2018-09-10:
for me pid_at_min_throttle = OFF is just one command line parameter and not two ;)

and the big advantage is pid_at_min_throttle = OFF describes exactly what mr. steel does in his video.

just to check if the motors hit something also can be done with a simple arm / disarm. i personally still do see not a advantage or need for this mode

borisbstyle, 2018-09-12:
I indeed also don't see a need for this mode as what @wind0r  already describes this simply can be achived with pid_at_min_throttle setting. It is actually designed for the same reason like you describe as people wanted this from KISS FC

closed 2018-10-17 by rephlex

wucke13, 2021-06-06:
I would like to recap this. I think what @rephlex desired (and what I would like as well) is an option to enable idle throttle on prearm.

I hope the following pseudo code makes it clear. 

```
if (feature_prearm_idle_throttle && mode_prearm && !mode_arm){
  for (i=0; i < number_of_motors; i++ )
    motor_throttle_signal[i] = idle_trottle;
}
```

Is this currently possible in betaflight?

My use case is a preflight check: when in this mode, one can easily spot all kind of issues with the motors and propellers (bent axis, interference with grass, bent blades, spin direction, etc.). And it's still rather safe to hold the copter by hand in this state, since nothing can make the motors accelerate - I at least much easier move the throttle accidentally than that I toggle my arm switch involuntarily. As this condition is only active when prearmed but not armed, it will under no conditions affect anything during flight.