betaflight/betaflight #6571: Broken sbus connectivity BF3.4/3.5 on revolt v1

neraja1, 2018-08-15:
Everything works untill Betaflight.3.3
On 3.4 / 3.4.1 / 3.5 i am losing connectivity with my reciever
Rx Frsky XSR
Board F4 revolt v1
rx is connected through sbus protocol and no telemetry used
Downgrading fixes the problem.
Any help

mikeller, 2018-08-15:
@proggod: I suspect that this is caused by the change to having REVOLT as a standalone target, and it probably just needs an adjustment in the configuration?

neraja1, 2018-08-17:
I am reading on #5669 as a workarround to close circuit tx pad and rx pad.
Do i understand correctly?
any chance to magic smoke from that?

jflyper, 2018-08-17:
@neraja1 
Can we see CLI `diff` outputs from working 3.3 configuration and non-working 3.5 configuration?
It is best if you can full-chip-erase-flash your board and load each diff to verify their operative status.

neraja1, 2018-08-17:
ok i have uploaded dump and diff from 3.3 (working) and 3.5 (not working)
on 3.5 i have set only my receiver configuration from the configurator , so no copy paste diff

[komos220 bf 3.5 dump.txt](https://github.com/betaflight/betaflight/files/2298382/komos220.bf.3.5.dump.txt)
[komos220 bf 3.5 diff.txt](https://github.com/betaflight/betaflight/files/2298383/komos220.bf.3.5.diff.txt)
[komos220 bf 3.3 dump.txt](https://github.com/betaflight/betaflight/files/2298384/komos220.bf.3.3.dump.txt)
[komos220 bf 3.3 diff.txt](https://github.com/betaflight/betaflight/files/2298385/komos220.bf.3.3.diff.txt)

neraja1, 2018-08-17:
and a diff and dump betaflight 3.5 (not working) with diff loaded
[komos220 bf 3.5 dump copied.txt](https://github.com/betaflight/betaflight/files/2298413/komos220.bf.3.5.dump.copied.txt)
[komos220 bf 3.5 diff copied.txt](https://github.com/betaflight/betaflight/files/2298414/komos220.bf.3.5.diff.copied.txt)

jflyper, 2018-08-18:
@neraja1 

Try this

```
set serialrx_halfduplex = OFF
```

neraja1, 2018-08-18:
@jflyper that fixed it , now it works like a charm. 1000 thanks :)

closed 2018-08-18 by neraja1

jflyper, 2018-08-18:
https://github.com/betaflight/betaflight/wiki/REVOLT-(V1-&-V2)#revolt-v1-sbus-configuration-after-v340