betaflight/betaflight #6573: Opentx telemetry lua scripts now working with BF 3.5

trojanGoat, 2018-08-16:
**Describe the bug**
Lua scripts stopped working after flashing 3.5.  Transmitter is still displaying telemetry data, but lua script is not working
**To Reproduce**
Flashed BF 3.5 on Emax magnum F3 using fport
Flashed BF 3.5 on Omnibus F4 v5 using fport

**Expected behavior**
LUA Scripts on taranis does not work

**Flight controller configuration**
# Betaflight / OMNIBUS (OMNI) 3.5.0 Aug 14 2018 / 03:49:05 (1d710afd7) MSP API: 1.40

name trojanGoat
resource SERIAL_TX 11 A08
resource TRANSPONDER 1 NONE

feature -RX_PPM
feature RX_SERIAL
feature MOTOR_STOP
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
serial 1 64 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200
aux 0 0 0 1275 2100 0 0
aux 1 27 2 1700 2100 0 0
aux 2 13 1 1700 2100 0 0
aux 3 26 0 1700 2100 0 0
aux 4 35 3 1700 2100 0 0
aux 5 36 2 1300 1700 0 0
rxfail 4 s 1000
set gyro_sync_denom = 2
set gyro_lowpass_hz = 120
set dyn_notch_quality = 15
set dyn_notch_width_percent = 30
set acc_hardware = NONE
set baro_hardware = NONE
set min_check = 1025
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set blackbox_device = NONE
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 0
set failsafe_switch_mode = KILL
set failsafe_throttle_low_delay = 0
set bat_capacity = 360
set current_meter = NONE
set beeper_dshot_beacon_tone = 3
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set tlm_inverted = ON
set tlm_halfduplex = OFF
set osd_cap_alarm = 400
set osd_craft_name_pos = 2091
set osd_warnings_pos = 2441
set osd_avg_cell_voltage_pos = 2476
set vtx_band = 5
set vtx_freq = 5658
set vcd_video_system = PAL
profile 0

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set smart_feedforward = ON
set f_pitch = 100
set f_roll = 100
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100
rateprofile 0

set roll_rc_rate = 80
set tpa_rate = 15
set tpa_breakpoint = 1050

**Setup / Versions**
 - Emax magnum F3
 - RXSR with fport firmware
 - using uninverted telemetry hack 

**Additional context**
Add any other context about the problem here.

jflyper, 2018-08-16:
@trojanGoat 
I see that you are using F.Port. Maybe related: #6565
- Is receiver firmware up to date? (#6565 fixed by updating)
- Do you see regular telemetry items?
- Did you try deleting/discovering sensors?

trojanGoat, 2018-08-16:
I can see all the values, vfas, A4, rssi, temperature and I can even set the channels on my vtx.  Just the betaflight settings won't show up on the taranis.

ghostface, 2018-08-16:
I have exactly the same problem.
F.port
omnibusf4 
r-xsr with R-XSR_FPORT_LBT180228 firmware
1.0.1 lua scripts

VTX menu works - pids do not.

mikeller, 2018-08-16:
Duplicate of #6565.

closed 2018-08-16 by mikeller

george-hawkins, 2020-12-30:
@19Roger53 - replying two years late on this on the off chance that you or someone else may be interested. I've been debugging the workings of the TX805 after discovering the behavior you describe.

**TLDR;** this is an Eachine issue and not a Betaflight issue. Eachine simply haven't implemented the SmartAudio 2.0 protocol correctly.

If you look at [`vtx_smartaudio.c`](https://github.com/betaflight/betaflight/blob/master/src/main/io/vtx_smartaudio.c), you can see that you set a mode for the VTX with `saSetMode`. To set the VTX to start up in normal pit mode one sends `saSetMode(SA_MODE_SET_IN_RANGE_PITMODE)` and then to turn it off one sends `saSetMode(SA_MODE_SET_IN_RANGE_PITMODE | SA_MODE_CLR_PITMODE)`.

This is how `vtxSASetPitMode` turns off pit mode. However, Eachine seem to have gotten things completely wrong. If you call `saSetMode(SA_MODE_SET_IN_RANGE_PITMODE)` then it turns on pit mode straight away (rather than simply configuring things such that the VTX starts up in pit mode next time you turn it on). And if you then try to turn off pit mode, using the OpenTX Lua script or the pit mode toggle in the Configurator, then it just ignores the `SA_MODE_CLR_PITMODE` flag and stays in pit mode.

If you don't have an OSD then your only option is to hold down the button on the VTX for 16 seconds to take it out of pit mode (which also means it will also not start up in pit mode next time you power cycle it).

If you do have an OSD then you can enter and exit pit mode using the `RACE` and `FREE` settings (as described [here](https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#config-sub-menu)). `RACE` and `FREE` _should_ set the VTX's startup behavior but on the TX805 they switch it in and out of pit mode immediately (once you select `SAVE` in the OSD).

So basically this is an Eachine issue - they simply ignore the `SA_MODE_CLR_PITMODE` flag. One could add hacks to `vtxSASetPitMode` for the TX805 (that one could enable via a new setting, e.g. `vtx_enable_tx805_pit_mode`) to allow you to immediately enter pit mode via with `saSetMode(SA_MODE_SET_IN_RANGE_PITMODE)` and leave it with `saSetMode(0)`. But I think it's impossible to support the situation where you can leave pit mode while still maintaining the setting that means that the VTX will start back into pit mode next time it's power cycled.

I'd say most people are using the TX805 with the OSD and always put the VTX into `RACE` mode before turning it off and then switch it into `FREE` mode next time they power it up (rather than being able to switch out of pit mode without permanently unsettling `RACE` mode).

As Eachine silently make changes to the same model over time, it wouldn't surprise me to hear that this behavior has changed between older and newer TX805 versions.

19Roger53, 2020-12-30:
GeorgeYes, I still fly 805's and have just resigned myself that sa would not work properly. I usually fly alone so it has not been an issue (but an annoyance) until this fall when I joined a club. So thanks for the information
be safe and Happy flying

george-hawkins, 2021-01-01:
Apologies to others for using a Betaflight issue for user-to-user messaging.

@19Roger53 - I've created a draft [PR](https://github.com/betaflight/betaflight/pull/10436) that compensates for the TX805 shortcomings and now allows me to switch in and out of pit mode from my TX.

The PR hasn't been terribly well received so far - for the understandable reason that people don't feel that Betaflight should be polluted with code to compensate for the incorrect behavior of devices like Eachine's TX805.

Anyway, if you'd be interested in trying out the Betaflight firmware with my fix then just tell me the target you'd like me to build, e.g. _IFLIGHT_H743_AIO_ or whatever, and the version you'd like to use e.g. 4.2.5. Either respond to this issue or you can find an email address to contact me at here - <https://gist.github.com/george-hawkins/92578e8a5bafce6f8500f5457ccbfbd2>