betaflight/betaflight #6574: CLI input not possible (DYS F4 PRO V2)

selectha, 2018-08-16:
**Describe the bug**
When I write something in CLI, it often does not work. Somethings I can enter a command sometimes it does not work. DUMP never shows me the complete list. It stops after about 3/4 and enters the last line in to the input field.
I am not sure if it is the configurator or the firmware.

**To Reproduce**
Steps to reproduce the behavior:
I have not tested a rollback to 3.4. The behavior is the same in Win10 and OSX with configurator 10.4.

**Expected behavior**
I should be able to enter any commands in to the input field.

**Flight controller configuration**
Diff an Dump not possible.

**Setup / Versions**
 - Flight controller DYS F4 PRO V2
 - Other components RX, VTX not yet attached.

mikeller, 2018-08-16:
@selectha:

>  DUMP never shows me the complete list.

This is indicative of a corrupt configuration. Do a `defaults`, then restore from backup, and it will work again.

selectha, 2018-08-16:
this worked. thank you very much mike!

closed 2018-08-16 by selectha