betaflight/betaflight #6577: Sensor lost and lua scripts bug

Pompon49, 2018-08-16:
Sensor lost warning all 10 seconds
Lua scripts just work for vtx remote

**To Reproduce**
Upgrade from 3.3.3 to 3.5:

**Expected behavior**
Lua scripts working and no sensor lost

**Flight controller configuration**
'''
diff

# version
# Betaflight / MATEKF405 (MKF4) 3.5.0 Aug 14 2018 / 03:43:51 (1d710afd7) MSP API: 1.40

board_name MATEKF405
manufacturer_id 

# name
name Eagle 1

# resources
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource SERIAL_TX 11 A08

# mixer

# servo

# servo mix

# feature
feature AIRMODE

# beeper
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -ON_USB
beeper -BLACKBOX_ERASE

# beacon

# map
map TAER1234

# serial
serial 1 0 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 30 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1100 0 0
aux 1 1 2 1150 1325 0 0
aux 2 2 2 1700 2100 0 0
aux 3 13 3 1375 1625 0 0
aux 4 35 3 1675 2100 0 0
aux 5 36 1 1825 2075 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 s 1500

# master
set gyro_lowpass_hz = 120
set mag_hardware = NONE
set baro_hardware = NONE
set rc_smoothing_type = FILTER
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 750
set motor_pwm_protocol = DSHOT600
set vtx_band = 1
set vtx_channel = 6
set vtx_power = 3
set vtx_freq = 5765

# profile
profile 0

set dterm_lowpass2_hz = 250
set dterm_notch_cutoff = 0
set iterm_relax = RP
set i_pitch = 75
set i_roll = 65

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set yaw_rc_rate = 140
set roll_expo = 45
set pitch_expo = 45
set yaw_expo = 20
set yaw_srate = 75

#  '''

**Setup / Versions**
 - Matek F405 OSD; Matex FCHUB 6S ; Matex vtx hv
 - R XSR With FPORT 
 - wired up as the mapping on matek site, just used motor 6 for softserial to use fport.

jflyper, 2018-08-16:
@Pompon49 
Did you consult #6565 and #6573?

jflyper, 2018-08-16:
@Pompon49 
Btw, code section separator is triple _**back**_ quotes.

Pompon49, 2018-08-16:
Ok #6565 solved a lot of issues i haven't seen it.
I 've just still have problem with Betaflight/PWM screen (lua) which still not work

jflyper, 2018-08-17:
@Pompon49 
Can you describe how PWM page is not working?

m013, 2018-08-20:
the bug  returned  after  i switched r-xsr for r9 mini (sbus+smartport via softserial).
i technically cannot use fport