betaflight/betaflight #6578: GPS Rescue - Flight mode OSD - not displayed unless current mode is AIR or related

yo8aiv, 2018-08-17:
Not shure if this is normal but 
GPS Rescue - Flight mode OSD - not displayed unless current mode is AIR or related

just updated from 3.4 (had the same issue but gps rescue was working) to 3.5
expected RESC to be displayed on OSD insteda of STAB or HOR but i saw it only shows  when in ACCRO/AIR modes. 

Gear: 
kakutev1 - board_name KAKUTEF4
rx - xsr 
gps - bn-180
frame - x215

if needed i can post a video or whatever is necessary.
Thanks.

yo8aiv, 2018-08-17:
yes by switch.

yo8aiv, 2018-08-17:
# diff

# version
# Betaflight / KAKUTEF4 (KTV1) 3.5.0 Aug 14 2018 / 03:39:46 (1d710afd7) MSP API: 1.40

board_name KAKUTEF4
manufacturer_id 

# name
name yo8aiv-X215

# resources
resource MOTOR 6 NONE
resource SERIAL_TX 11 C08

# mixer

# servo

# servo mix

# feature
feature SOFTSERIAL
feature GPS
feature LED_STRIP

# beeper

# beacon

# map
map TAER1234

# serial
serial 5 2 115200 57600 0 115200
serial 30 2048 115200 57600 0 115200

# led
led 0 6,8::COI:13
led 1 6,9::COI:13
led 2 9,9::COI:13
led 3 9,8::COI:13

# color

# mode_color

# aux
aux 0 0 0 1900 2100 0 0
aux 1 1 5 900 1100 0 0
aux 2 2 5 1350 1650 0 0
aux 3 27 2 1800 2100 0 0
aux 4 46 6 1800 2100 0 0
aux 5 13 4 1700 2100 0 0
aux 6 15 4 1475 2100 0 0
aux 7 19 1 1300 1700 0 0
aux 8 31 1 1700 2100 0 0
aux 9 35 7 1700 2100 0 0
aux 10 36 3 1600 2100 0 0
aux 11 47 5 1800 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 10 s 2000

# master
set rssi_channel = 16
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 20
set failsafe_throttle = 1255
set failsafe_procedure = AUTO-LAND
set bat_capacity = 1500
set current_meter = ADC
set ibata_scale = 300
set ibata_offset = -3
set yaw_motors_reversed = ON
set gps_provider = UBLOX
set gps_rescue_descent_dist = 100
set gps_rescue_ground_speed = 1000
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1500
set osd_alt_alarm = 150
set osd_vbat_pos = 2455
set osd_rssi_pos = 2114
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2102
set osd_flymode_pos = 2274
set osd_g_force_pos = 2392
set osd_throttle_pos = 2241
set osd_vtx_channel_pos = 2135
set osd_ah_pos = 2126
set osd_current_pos = 2400
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 361
set osd_gps_speed_pos = 2296
set osd_gps_lon_pos = 2049
set osd_gps_lat_pos = 2064
set osd_gps_sats_pos = 2081
set osd_home_dir_pos = 2126
set osd_home_dist_pos = 2093
set osd_altitude_pos = 2230
set osd_pidrate_profile_pos = 2169
set osd_avg_cell_voltage_pos = 2444
set osd_battery_usage_pos = 360
set osd_disarmed_pos = 2314
set osd_nvario_pos = 2264
set osd_rtc_date_time_pos = 5
set osd_core_temp_pos = 2424
set osd_stat_rtc_date_time = ON
set osd_stat_max_dist = ON
set osd_stat_endbatt = ON
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set vcd_video_system = NTSC
set displayport_max7456_wht = 1

# profile
profile 1

set dterm_notch_cutoff = 0
set vbat_pid_gain = ON

# rateprofile
rateprofile 3

jflyper, 2018-08-17:
@yo8aiv 
Okay, the issue is reproducible and I now know why it is happening.

jflyper, 2018-08-17:
The current code is
```
            if (FLIGHT_MODE(FAILSAFE_MODE)) {
                strcpy(buff, "!FS!");
            } else if (FLIGHT_MODE(ANGLE_MODE)) {
                strcpy(buff, "STAB");
            } else if (FLIGHT_MODE(HORIZON_MODE)) {
                strcpy(buff, "HOR ");
            } else if (FLIGHT_MODE(GPS_RESCUE_MODE)) {
                strcpy(buff, "RESC");
            } else if (IS_RC_MODE_ACTIVE(BOXACROTRAINER)) {
                strcpy(buff, "ATRN");
            } else if (isAirmodeActive()) {
                strcpy(buff, "AIR ");
            } else {
                strcpy(buff, "ACRO");
            }
```

If `GPS_RESCUE_MODE` has a display precedence over other modes, then checking for this mode should precede the "other" modes.

Is it something like this?
1. FS
2. RESCUE
3. ANGLE, HORIZON, ATRAINER
4. AIR
5. ACRO

etracer65, 2018-08-17:
@jflyper The proposed reordering looks reasonable to me.

jflyper, 2018-08-17:
Oh, wait. RESCUE works as FS 2nd stage, doesn't it?

etracer65, 2018-08-17:
Yes, in that case both `FLIGHT_MODE(FAILSAFE_MODE)` and `FLIGHT_MODE(GPS_RESCUE_MODE)` will be true.

Although I'm not sure that the GPS Rescue message should override the failsafe message. The failsafe is the important part.  Take the case where the user is not using the GPS-RESCUE failsafe_procedure.  If they manually activate the mode it should display "RESCUE" but if a failsafe occurred we wouldn't want the "RESCUE" to override the "!FS!" otherwise there would be no way to tell a failsafe occurred.  But that's just my opinion.

PR #6580 (2018-08-20): OSD GPS RESCUE has display precedence over other modes except FS
Fixes #6578 

Should also be applied to 3.5.x maintenance tree.
closed 2018-08-20 by mikeller