betaflight/betaflight #6579: Sharing port (Telemetry and Serial RX)

0Jiahao, 2018-08-17:
Hi, is it possible to use one UART port for both telemetry output and serial rx?

jflyper, 2018-08-17:
It depends on serial rx protocol you are using.

0Jiahao, 2018-08-17:
I am using the MAVLink Telemetry.

mikeller, 2018-08-18:
@0Jiahao: @jflyper's statement is incorrect. You can share an UART between MAVLink and a serial RX protocol, as long as the serial RX protocol only requires a simplex RX connection to the flight controller:

https://github.com/betaflight/betaflight/blob/98616d469012e5287f35c9302bbea70d1e8b920d/src/main/rx/fport.c#L237-L2telemetry/telemetry.h39

The only restriction there is that the MAVlink data will be sent at the same baud rate that is used by the serial RX protocol that you are using.

0Jiahao, 2018-08-19:
Thanks for your replies. I modified the src/main/telemetry/mavlink.c to enable my customized function (output telemetry and receive the command from external processer), and it works.

mikeller, 2018-08-19:
@0Jiahao: What kind of external processor are you using? It might be something worth contributing back to the community.

jflyper, 2018-08-19:
@0Jiahao @mikeller Apologies for incorrect information. I was assuming S.BUS which runs at non-standard 100Kbps, so TX side is usually not compatible with telemetry protocols.

0Jiahao, 2020-07-30:
> @0Jiahao: What kind of external processor are you using? It might be something worth contributing back to the community.

It was [Jevios](http://jevois.org/)

mikeller, 2020-08-01:
@0Jiahao: Nice, sounds interesting. Did anything come from your experimentation?