betaflight/betaflight #6582: Average Cell voltage on crossfire telemetry

thomasbridi, 2018-08-17:
So far the CRSF telemetry shows just the main battery voltage. However, many pilots started to use quadcopters with different numbers of cells 3-4-5-6s, moreover, many builds can work with multiple different batteries. Thus, the main battery voltage (in my opinion) is not enough. The average cell voltage is more useful. I don't know if it is possible to add this variable to the CRSF telemetry without any change in the CRSF firmware. However, I think should be easy to change the main battery voltage with the average cell voltage. So my suggestion is to add the average cell voltage to the CRSF telemetry or, if that is not possible, add the possibility to select between the main battery voltage and the average cell voltage.

mikeller, 2018-08-18:
This could be done in the same fashion as it is done for SmartPort / FrSky hub / iBus, where `telemetryConfig()->report_cell_voltage` can be used to switch the voltage sensor to reporting the average per cell voltage.

thomasbridi, 2018-08-20:
in `crsfFrameBatterySensor() `i've put 
`if (telemetryConfig()->report_cell_voltage)
    {
        sbufWriteU16BigEndian(dst, getBatteryVoltage()); // vbat is in units of 0.1V
    }
    
    else
    {
        sbufWriteU16BigEndian(dst, getBatteryAverageCellVoltage());
    }`
and then in betaflight I've set `report_cell_voltage = ON`

but it is not working

mikeller, 2018-08-20:
@thomasbridi: Hmm, looks like you've put it the wrong way round - I bet it's working if you set `report_cell_voltage = off`. :wink:

thomasbridi, 2018-08-22:
Yes it worked. LOL. Can I commit it?

thomasbridi, 2018-08-22:
Obviously I've fixed the mistake

mikeller, 2018-08-22:
@thomasbridi: Yes, please open a pull request for this.

thomasbridi, 2018-08-22:
i'm following the steps in docs/development.md but at step 9 it says:
remote: Permission to betaflight/betaflight.git denied to thomasbridi.
fatal: unable to access 'https://github.com/betaflight/betaflight.git/': The requested URL returned error: 403

mikeller, 2018-08-22:
@thomasbridi: Looks like you went wrong somewhere around 

> 1. Login to github, go to the cleanflight repository and press fork

In the next step, you need to clone **your fork** (i.e. **not** `betaflight/betaflight.git`, but `thomasbridi/betaflight.git`.

thomasbridi, 2018-08-23:
ok, I've found that the IMPULSERCF3 board does overflow the flash, how can I solve this? 
Is there a define that states if the board is IMPULSERCF3 so as I can disable this feature for this board? I did not found it

thomasbridi, 2018-08-23:
maybe:
`#if defined(STM32F3)
    sbufWriteU16BigEndian(dst, getBatteryVoltage());
#else
    if (telemetryConfig()->report_cell_voltage) {
        sbufWriteU16BigEndian(dst, getBatteryAverageCellVoltage()); // vbat is in units of 0.1V
    } else {
        sbufWriteU16BigEndian(dst, getBatteryVoltage());
    }
#endif`
is this ok?

mikeller, 2018-08-23:
@thomasbridi: Try starting off with a fresh branch off master as it is right now, now it should fit.

mikeller, 2018-08-23:
@thomasbridi: I just saw that you've opened your pull request against cleanflight. We've removed LED_STRIP from F3 in Betaflight during the development phase to make some room, try opening your pull request against master in Betaflight, it should fit there.

But I've also seen that tests are failing. You'll have to stub out `getBatteryAverageCellVoltage()` in `src/test/unit/telemetry_crsf_msp_unittest.c` in order to make it work.

k4cus, 2021-09-10:
It can be done easily by using this lua script:
https://github.com/k4cus/Lua-scripts/blob/main/Vcell.lua

tavdog, 2022-07-23:
Close ?

haslinghuis, 2022-07-23:
Fixed in https://github.com/betaflight/betaflight/pull/6624

closed 2022-07-23 by haslinghuis