betaflight/betaflight #6583: Why do you guys not have betaflight 3.5 firmware for the CL_Racing F4 flight controller because I bought it thinking I would get all the updates and now I am a little upset because now my firmware is out of date.

Ekoran, 2018-08-17:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

DieHertz, 2018-08-17:
It's called CLRACINGF4, rings any bells? :)

watageek, 2018-08-17:
[image]

Have a closer look...

jflyper, 2018-08-18:
## @Ekoran 

# You filed the exact same issue 15 days ago and people wrote the exact same answer.

## Ref: https://github.com/betaflight/betaflight/issues/6512