betaflight/betaflight #6584: Cannot change OSD font on DYSF4Pro Flight Controller

Doobie52, 2018-08-18:
I just upgraded to 3.5 and after uploading the chosen font, it reboots back to default.  Has anyone else had this problem using this board?

mikeller, 2018-08-18:
@Doobie52: Are you actually checking in the OSD, or just in configurator? Configurator does not have a way to know what font is installed on the flight controller, and will always show the default font when opening the tab.

Doobie52, 2018-08-18:
Yes, I have my monitor right next to the computer as I'm setting up the OSD so I can position everything where I want it on the screen.

mikeller, 2018-08-18:
@Doobie52: Looks like the MAX7456 chip on your board is a dud then. Get a warranty replacement from your manufacturer / supplier.

closed 2018-08-18 by mikeller

Doobie52, 2018-08-19:
So I just reflashed to 3.5 again and I can change fonts now.  Go figure.  Thanks for the input.