betaflight/betaflight #6587: Yaw Feedforward resetting to 0

Hammerburger12, 2018-08-18:
I tried setting the feedfoward value to 100 as per the recommendation in the release info, but no matter what value i set it too every time i save and refresh its reset itself back to 0. Would anyone know the reason for this and if there is anything i have to change to make it stick?

mikeller, 2018-08-18:
@Hammerburger12: If you had followed the instructions and done what you were asked to do when opening an issue, and posted the information that you were asked to provide, there would be a better chance of somebody being able to help you.

Just making guesses here, but you probably have RC interpolation for yaw disabled.

etracer65, 2018-08-20:
Closing for lack of response.

closed 2018-08-20 by etracer65

brandonsD, 2018-09-08:
Come on now. I have this same issue. In BetaFlight 3.5 my feed forward resets to zero when I go back into the cli and I tried to change it and save but it just goes back to zero. What’s the deal with that?

brandonsD, 2018-09-08:
I’m using f3 light flight controler. BetaFlight config 10.4.0 with FW BTFL3.5.0 target srf3

brandonsD, 2018-09-08:
Also throttle boost resets to zero as well. What’s that about??

etracer65, 2018-09-08:
The answer was given above. You need to have smoothing enabled for roll, pitch and yaw for feedforward to work.  It must be enabled on throttle for throttle boost to work.  See the Receiver tab.

goodvibes, 2018-10-27:
Perhaps if the initial answer wasn't unnecessarily rude and obfuscated it would be more helpful for resolving the problem.

In plain English: On the receiver tab, under RC Smoothing, ensure RPYT is selected to enable FF on Yaw and Throttle Boost.

The default is RPT, which should perhaps be the actual bug report if it's causing confusion. If an option is unavailable it should be disabled with a mouse-over providing information why. To have silent resets is terrible practice.

etracer65, 2018-10-28:
Firstly, the default is RPYT. Secondly if you have ways to improve the user experience then contribute code rather complain.

mikeller, 2018-10-28:
@goodvibes: I don't think that pointing out to somebody that there is only little chance that anybody will be able to help them with their support request because, despite being explicitly asked for it, they failed to provide the bare minimum of information that is required is rude - it's just a statement of facts.

Also, as I clearly state, due to lack of information, my diagnosis is just a wild guess. No point in providing details for something that may not even be a fix for the problem.

And yes please, we'll be grateful for everybody who volunteers to sink some time into UX improvements (or any other improvements).