betaflight/betaflight #6590: Matek F722. Accelerometer not responding in calibration after 4.0.0 #1055 push.

Victek, 2018-08-18:
Description:
Accelerometer can't be calibrated, quad becomes unstable after 30 sec flight (angle mode).

Steps to reproduce:
Upgrade firmware from 3.5.1 to 4.0.0 and try to calibrate gyro/acc.

Diff.
In 3.5,1
```
# version
# Betaflight / MATEKF722 (MKF7) 3.5.1 Aug 15 2018 / 21:24:38 (2f6053d17) MSP API: 1.40
board_name MATEKF722
manufacturer_id 
# name
name Mart4-220
# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 B01
# mixer
# servo
# servo mix
# feature
feature MOTOR_STOP
# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
serial 0 8192 115200 57600 0 115200
# led
# color
# mode_color
# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1800 2100 0 0
aux 3 13 2 1800 2100 0 0
aux 4 28 3 1300 1700 0 0
aux 5 35 3 1800 2100 0 0
aux 6 47 4 1300 1700 0 0

# adjrange
# rxrange
# vtx
# rxfail
# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set acc_calibration = 23,37,16
set mag_hardware = NONE
set min_check = 1015
set max_check = 2000
set fpv_mix_degrees = 15
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set bat_capacity = 1500
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 36
set vbat_scale = 109
set ibata_scale = 350
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1000
set osd_vbat_pos = 2499
set osd_rssi_pos = 2104
set osd_tim_1_pos = 469
set osd_tim_2_pos = 2517
set osd_flymode_pos = 2083
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_craft_name_pos = 2506
set osd_pidrate_profile_pos = 2488
set osd_warnings_pos = 2090
set osd_avg_cell_voltage_pos = 2467
set vtx_band = 2
set vtx_channel = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5790
set vtx_pit_mode_freq = 5790
set vcd_video_system = PAL

# profile
profile 0
# rateprofile
rateprofile 0 
```

In 4.0.0 #1055
```
# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Aug 16 2018 / 13:13:44 (968583710) MSP API: 1.40
board_name MATEKF722
manufacturer_id 
# name
name Mart4-220
# resources
resource MOTOR 5 NONE
resource CAMERA_CONTROL 1 B01
# mixer
# servo
# servo mix
# feature
feature MOTOR_STOP
# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED
# beacon
beacon RX_LOST
beacon RX_SET
# map
map TAER1234
# serial
serial 0 8192 115200 57600 0 115200
# led
# color
# mode_color
# aux
aux 0 0 0 1850 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 2 1 1800 2100 0 0
aux 3 13 2 1800 2100 0 0
aux 4 28 3 1300 1700 0 0
aux 5 35 3 1800 2100 0 0
aux 6 47 4 1300 1700 0 0
# adjrange
# rxrange
# vtx
# rxfail
# master
set gyro_sync_denom = 2
set gyro_use_32khz = ON
set acc_calibration = 22,32,14
set mag_hardware = NONE
set min_check = 1015
set max_check = 2000
set fpv_mix_degrees = 15
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set dshot_idle_value = 300
set motor_pwm_protocol = DSHOT600
set failsafe_switch_mode = STAGE2
set bat_capacity = 1500
set vbat_min_cell_voltage = 34
set vbat_warning_cell_voltage = 36
set vbat_scale = 109
set ibata_scale = 350
set beeper_dshot_beacon_tone = 4
set small_angle = 180
set deadband = 1
set yaw_deadband = 1
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_cap_alarm = 1000
set osd_vbat_pos = 2499
set osd_rssi_pos = 2104
set osd_tim_1_pos = 469
set osd_tim_2_pos = 2517
set osd_flymode_pos = 2083
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_craft_name_pos = 2506
set osd_pidrate_profile_pos = 2488
set osd_warnings_pos = 2090
set osd_avg_cell_voltage_pos = 2467
set vtx_band = 2
set vtx_channel = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5790
set vtx_pit_mode_freq = 5790
set vcd_video_system = PAL
# profile
profile 0
# rateprofile
rateprofile 0
```

Setup / Versions
 - Matek F722 working fine with 3.5.1, problem arised after 4.0.0 #1055 pushed.
 - Frysky F.Port (R-XSR), Racestar 4in1 30A, Emax 2306, 4S.

jflyper, 2018-08-18:
@Victek 
Reproduced with 968583710 on multiple targets including MATEKF722(F722), NERO(F722), OMNIBUSF7V2(F745), OMNIBUSF4(F405).

However, #1055 is over 2 years old. Can you elaborate?

mikeller, 2018-08-18:
@jflyper: I think think ##1055 is a reference to the CI build.

etracer65, 2018-08-19:
Seems more likely to be related to https://github.com/betaflight/betaflight/pull/6536 which also ended up in the same build.

jflyper, 2018-08-19:
Yeah. Bisecting also blamed #6536.

jflyper, 2018-08-19:
@leocb Can you take a look?

leocb, 2018-08-19:
I will, thanks for the warning

AirBreak69, 2018-08-19:
@leocb @jflyper @etracer65 
acceleration.c line 365: acc.dev.acc_1G_rec = **_(float)_** 1 / acc.dev.acc_1G;
Here a (float) cast is needed. Otherwise acc.dev.acc_1G_rec will be 0.0
Tested it looking at the OSD: G-Force comes 1.0G again instead of 0.0G.

DieHertz, 2018-08-19:
No float casts, just a float literal ;)

jflyper, 2018-08-19:
# `1.f`

whoFPV, 2018-08-19:
Had a similar issue. Can you confirm you’re getting gyro response as well? In my issue (still unresolved) I flashed 3.4 and greyed out accel calibration and gyro loss upon third save & reeboot. Tried to backtrack and no luck. Target DYSF4PRO 10.4 config

whoFPV, 2018-08-19:
@airkbreak69 I have the dys mini f4. And it’s strange as it’s happened to two fc’s back to back. Scared to flash again as I have two other builds with same setup. Waiting to hear from manufacturer as well as it may be something on their end before doing any updating. 4.0 will have to wait :(

closed 2018-08-20 by mikeller

Victek, 2018-08-20:
Tested in Matek F722, solved, Thank you!

PR #6598 (2018-08-20): Fixed problems with accelerometer.
Fixes #6590.