betaflight/betaflight #6592: MATEK F722.  SBUS not working with FrSky R9 Slim+ receiver

BatesKG, 2018-08-18:
**Describe the bug**
Connecting an R9 Slim+ receiver to a Matek F722 FC,   
S.port telemetry is showing altitude from the onboard barometer on Horus s10s (FrOS), but no receiver response is showing in the Betaflight Rx tab.  SBUS is set up correctly in the Ports and Configuration tabs.  I tried the oldest and newest versions of the R9 Slim+ / R9M fireware....same result.

**To Reproduce**
Just connect an R9 Slim+ Rx to an F722 FC.  I tried two different receivers.  I also tried BF 3.4.1, and it showed the same behavior.  

**Expected behavior**
SBUS to work.

**Flight controller configuration**

```
# diff

# version
# Betaflight / MATEKF722 (MKF7) 3.5.0 Aug 14 2018 / 03:45:13 (1d710afd7) MSP API: 1.40

board_name MATEKF722
manufacturer_id 

# name
name Roo-6

# resources

# mixer

# servo

# servo mix

# feature
feature GPS
feature RSSI_ADC
feature AIRMODE

# beeper

# beacon

# map

# serial
serial 0 2 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 4 8192 115200 57600 0 115200

# led
led 0 6,6::CTW:0
led 1 6,7::CTW:0
led 2 6,8::CTW:0
led 3 6,9::CTW:0
led 4 9,6::CTW:0
led 5 9,7::CTW:0
led 6 9,8::CTW:0
led 7 9,9::CTW:0

# color

# mode_color
mode_color 6 0 2
mode_color 6 1 0
mode_color 7 0 5

# aux
aux 0 0 3 900 1200 0 0
aux 1 1 0 1300 1700 0 0
aux 2 2 0 900 1200 0 0
aux 3 13 8 1300 1700 0 0
aux 4 19 8 900 1200 0 0
aux 5 26 3 900 1200 0 0
aux 6 39 9 1800 2100 0 0

# adjrange
adjrange 0 0 10 900 2100 12 10 0 0

# rxrange

# vtx

# rxfail
rxfail 12 s 1500
rxfail 13 s 2000

# master
set align_gyro = CW0
set align_acc = CW0
set align_mag = CW0
set mag_hardware = NONE
set max_check = 1950
set rc_interp_ch = RP
set rc_smoothing_input_type = PT1
set rc_smoothing_derivative_type = OFF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 9
set small_angle = 80
set gps_provider = UBLOX
set gps_sbas_mode = WAAS
set gps_auto_baud = ON
set deadband = 1
set yaw_deadband = 1

# profile
profile 0

set throttle_boost = 0
set i_pitch = 30
set d_pitch = 25
set p_roll = 40
set i_roll = 30
set d_roll = 20
set p_yaw = 60
set i_yaw = 35
set d_yaw = 30
set f_yaw = 0
set i_level = 0
set d_level = 0

# rateprofile
rateprofile 0
```

**Setup / Versions**
 - Flight controller Matek F722
 - Other components 
RX - R9 Slim+, 
IRC TrampHV

**Additional context**
Was working fine with a standard 2.4GH receiver using S.port.   R9 Slim+ is the only difference.

jflyper, 2018-08-19:
@BatesKG Have you had this particular receiver instance work in the past?

BatesKG, 2018-08-19:
No.   First attempt to use an R9 series Rx in Betaflight.  I have used other 2.4Ghz receivers successfully on this particular FC board though.
My x10s came with an R9M module so I thought I'd try a long range receiver in one of my builds.   Any of the other standard range receivers I've bought just connected fine via SBUS.  
After having the problem, I was doing some digging and found iNav Github Issue#3185 where the R9 series of Rx wasn't working in iNav 2.0 but was working in ver1.9.1.   I thought this may be something similar???

jflyper, 2018-08-19:
iNavFlight/inav#3185 was closed unconditionally.

There is another one; iNavFlight/inav#2827 < TL;DR 😬

BatesKG, 2018-08-19:
Sooooo...... what's the timing for SBUS in Betaflight?   I tried flashing back to ver 3.2.3 and still no luck.   According to the link you posted above, in iNav, SBUS worked with 3ms timing and broke when they implemented 6ms.    
The fact that other SBUS receivers work just fine on this FC in Betaflight, it must be something more specific to the R9 Slim series?

mikeller, 2018-08-19:
Maximum frame length has always been 3 ms in Betaflight:

https://github.com/betaflight/betaflight/blob/bbf493ebaee6958c3fd86ad472f691d79811254d/src/main/rx/sbus.c#L59

I haven't tested much with SBus, but the FPort firmware for R9 slim+ is working fine with Betaflight.

jflyper, 2018-08-21:
@BatesKG Search the next with "Betaflight RSSI OSD".

BatesKG, 2018-08-22:
I did search.  
After doing some _more_ searching, I found one solution.... set the rssi_scale to 200.   That doesn't really seem right, but it works.

BatesKG, 2018-08-22:
Thanks mikeller.  That's what I did before my last post and that seems to work.

So, just the original SBUS issue left.   Seeing that was the original bug reported, I'm assuming this report should stay open?  Or is using f.port a valid solution and the thread should be closed?   (this is my first submitted issue on Github)
Thanks to everyone for their help!

BatesKG, 2018-08-24:
Will do.   Thanks.

BatesKG, 2018-09-08:
Well....I purchased another Matek F722 FC and wired it up with an R9 harness.....and SBUS worked fine.    So there must have been some kind of internal failure on my first board that only broke SBUS?   

I tested with the new FC that SBUS worked with the R9 Slim+ on both the original firmware (3.2.2) as well as 3.5.0.   Worked fine.    
As this turned out to be hardware related, I will close this issue.   Thanks to everyone that contributed.

closed 2018-09-08 by BatesKG