betaflight/betaflight #6595: HID emulation feture not working properly

Azuxul, 2018-08-19:
When using HID emulation feature with Windows 10, in the controller device panel commands is good, but in software like simulation software (FPV Frerider, RDS ...) commands are not good (https://i.azuxul.fr/MeUtB3.mp4 and https://i.azuxul.fr/lrLrAm.mp4).
At the center of the real stick, sowtware show low stick. When stick goes down it go up on the software.

I test this feature with two FCs (FlameF4 (Target OBF4) and Matek F405-AIO) and two receiver types. In every case the problem is always here.

I test with BF 1.4.1 and 1.5.

mikeller, 2018-08-19:
This seems to be a problem of how these simulations are interpreting the values sent by the HID device in Windows. I tested FPV Freerider on linux with the HID device, and after calibration it's working just fine.

I am no expert in Windows game development, and how input from a HID device is supposed to be scaled for it. Somebody else will have to pick this up.

vodka-bears, 2018-09-03:
Also Liftoff in linux (steam, xubuntu 16.04, 18.04) doesn't detect my FC (OMNIBUSF4SD). System does. In windows everything works right.

mikeller, 2018-09-04:
@vodka-bears: This seems to be a problem of the application - the operating system properly recognises the device, and initialises a device driver for it. You'll have to contact the application's developer to get this fixed.

Yamaford, 2018-10-20:
I think we need to make a special model for freerider on the taranis and select whatever bind slot you have for that quad on that model then set up the taranis like the doc in freerider say. https://drive.google.com/file/d/0BwSDHIR7yDwSOGhPMTFyWEpDVWM/view?usp=sharing

Yamaford, 2018-10-20:
I'll try it tonight if I get a minute.

haslinghuis, 2022-04-03:
FPVFreerider Recharged works great here with Taranis X9D+ ACCST on Windows 10

https://itch.io/profile/haslinghuis.

closed 2022-04-03 by haslinghuis