betaflight/betaflight #6602: hot motor betaflight 3.5

eligys, 2018-08-20:
Hi when I flash betaflight 3.5 and go for test at the first moment when I start flight I hear strange noise from the motors after land I check motor and it was hot. trying put new prop and still get same problem. So I decide flash again betaflight 3.4 to check and it was fine again. I'm using kakute f4 FC and racestar br3605 2600kv motors.

mikeller, 2018-08-20:
@eligys: Have you read and applied the tuning tips for 3.5 (https://github.com/betaflight/betaflight/wiki/3.5-tuning-notes)?

dadoribeiro, 2018-08-20:
Be sure to use the latest configurator and do not just paste a diff. Configure all the filters as per wiki and start with defaults. Also try 8k gyro and 4K PID loop.

eligys, 2018-08-22:
I was try from defaults and get this problem, only what I change was me rates

closed 2018-09-28 by stale[bot]