betaflight/betaflight #6604: OMNINXT7 (ONX7) Hang up at DIFF and DUMP on CLI

RobMcFly, 2018-08-20:
Hello,

flashed to the OMNIBUS F7 NXT to Betaflight / OMNINXT7 (ONX7) 3.5.0 Aug 14 2018 / 03:51:17
Flying ist great.

But in CLI it´s not possible to get a complete DIFF (something 3 or 4 Lines will be shown) 
on DUMP nothing happens and the board hangs up. no reaction. need to reconnect. 
After reconnection it´s possible to configure everything. "Get" works.
Thanks in advance

aguindehi, 2018-08-20:
Dumping and diffing works fine on the two OMNINXT7 that we have here with Betaflight / OMNINXT7 (ONX7) 3.5.0 Aug 14 2018 / 03:51:17 (1d710afd7) MSP API: 1.40 (eg. the same version as you have).

jflyper, 2018-08-20:
@RobMcFly Try CLI "defaults" and start configuring again.

RobMcFly, 2018-08-21:
started with "defaults" = same
Flashed FW again = same

Tried another 3 quads with BF 3.5 works fine on them. Some wrong with the board itself? When I type "get" in CLI the last line comes out is "VCD_Video_system = PAL and it hang up. (When I click on another tab it´s "Waiting for data" Must reconnect then too. I get another OMNINXT7 later the day. I will experiment with that. Thanks guys

Mayk-CLP, 2018-08-21:
It´s working on mine without any issues. Do you have Beta Config 10.4.0?

RobMcFly, 2018-08-21:
Yes 10.4.0
Got another OMNINXT7 Board here now. That´s working fine. Hard to believe it´s an hardware issue, when everything else is working.

RobMcFly, 2018-08-21:
got these lines when entering cli:
$M>ni$M>$iÜÜÜuõÜÜÜÜÜÜÜÜÜÜÜÜÜË
Entering CLI Mode, type 'exit' to return, or 'help'

is that helpful?

mikeller, 2018-08-21:
@RobMcFly: That's normal, an artefact of the flight controller switching from MSP to CLI.

snotabe, 2018-08-23:
I have the same problem with omninxt7 after flashing BF 3.5 with configurator 3.4.0.
I can not make a dump anymore, it hangs and I have to reconnect to get connected.
I can not flash a new firmware / when the firmware has loaded online and I press to flash, nothing happens besides it looses serial connection.

I can configure whet I want through betafligt configurator, but something has happend.

jflyper, 2018-08-24:
@snotabe 
I think you have multiple different problems which are either hardware or configuration related.

Please get help from other user support options such as asking the manufacturer of the hardware you are using, RCGroups: https://rcgroups.com/forums/showthread.php?t=2464844, or Slack (registration at https://slack.betaflight.com/) or other user support forums & groups (e.g. Facebook).

That being said, try removing or powering down something connected to RX1 or RX3 (e.g., a RX receiver). Chatty devices interfere with boot loader (a DFU software) and causes your flashing to fail.
For further support, refer to above statement.

closed 2018-08-24 by jflyper