betaflight/betaflight #6608: Missing table entries in defaultAdjustmentConfigs table

pkruger, 2018-08-21:
**Describe the bug**
Missing entries in the defaultAdjustmentConfigs table in rc_adjustments.c for the following adjustment functions:
ADJUSTMENT_ROLL_RC_RATE,
ADJUSTMENT_PITCH_RC_RATE,
ADJUSTMENT_ROLL_RC_EXPO,
ADJUSTMENT_PITCH_RC_EXPO,

**To Reproduce**
N/A

**Expected behavior**
There should be an entry for all adjustment functions, otherwise later when a new function is added the new function's entry will not line up with the index into the table and the adjustment will not work correctly/not work at all.

**Flight controller configuration**
N/A

**Setup / Versions**
Betaflight 3.5.0

**Additional context**
The missing entries will cause the entries for:
    ADJUSTMENT_PID_AUDIO,
    ADJUSTMENT_PITCH_F,
    ADJUSTMENT_ROLL_F,
    ADJUSTMENT_YAW_F,
Not to align and not to work.

etracer65, 2018-08-21:
This is not a problem and the referenced adjustments work as expected.

The order of the entries in the `defaultAdjustmentConfigs` array is not positionally aligned with the enumeration of adjustments defined in rc_adjustments.h.  Instead one element of each entry in the array is a reference to which adjustment enumeration index the entry is for.

etracer65, 2018-08-21:
While I don't disagree that the code can be improved, it's not actually broken at the moment.  The line you're referring to is actually part of a block like this:
```
        if (isRangeActive(adjustmentRange->auxChannelIndex, &adjustmentRange->range) &&
            (adjustmentRange->adjustmentCenter == 0)) {
            const adjustmentConfig_t *adjustmentConfig = &defaultAdjustmentConfigs[adjustmentRange->adjustmentFunction - ADJUSTMENT_FUNCTION_CONFIG_INDEX_OFFSET];
            configureAdjustment(adjustmentRange->adjustmentIndex, adjustmentRange->auxSwitchChannelIndex, adjustmentConfig);
        }
```
The line you're concerned about is never actually reached for the unused entries at the end of the array because `isRangeActive()` returns false.

etracer65, 2018-08-21:
But I'll look into improving this when I get a chance.

pkruger, 2018-08-22:
Ah, ok, I understand what you mean.  I'm not criticising the code at all, anything can be improved given  enough time and money. :o)  I ran into the above issue recently while implementing feature request https://github.com/betaflight/betaflight/issues/4155 and adding an entry to the end of the defaultAdjustmentConfigs table which is a select adjustment.  I'll close this issue, I can add dummy entries in my case (feature) to make it work.

closed 2018-08-22 by pkruger