betaflight/betaflight #6626: connect OMNIBUSF4SD with HC-SR04

0Jiahao, 2018-08-23:
Hi, I am trying to connect the OMNIBUSF4SD with HC-SR04. I have connected trigger with pwm 5 and echo with pwm 6. I also modified the configuration via the CLI (following is my #diff). But the sonar is not detected, any idea why? 

I know the altitude hold is removed, but I still need the sonar sensor works.

=============================================================================
# diff

# version
# Betaflight / OMNIBUSF4SD (OBSD) 3.5.0 Aug 21 2018 / 17:11:59 (935733ee7) MSP API: 1.40

board_name OMNIBUSF4SD
manufacturer_id 

# name

# resources

# mixer

# servo

# servo mix

# feature
feature RANGEFINDER

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set rangefinder_hardware = HCSR04

# profile
profile 0

# rateprofile
rateprofile 0

jflyper, 2018-08-23:
`rangefinder_hardware = HCSR04`

jflyper, 2018-08-23:
@0Jiahao Sorry, range finder is completely disabled as of 3.4.0.

closed 2018-08-23 by jflyper

0Jiahao, 2018-08-23:
@jflyper thanks for your reply! and any idea which is the latest version that enables the rangefinder?

jflyper, 2018-08-24:
3.3.3