betaflight/betaflight #6628: sam m8q gps use

CcLaFleche, 2018-08-23:
Hi,
i use a SAM m8q GPS sensor, but he have no additionnal memory and return in stock settings communication after power down.

Betaflight can configure him at every start ?

jflyper, 2018-08-24:
Yes.

closed 2018-08-24 by jflyper

mikeller, 2018-08-24:
@CcLaFleche: According to my (limited) knowledge of GPS modules, there is a difference between differently priced ones: Good and expensive ones have on-board non-volatile storage that allows you to change settings and then persist them across reboots. Cheap ones don't, and have to be reconfigured on every boot.
Betaflight focuses on flight performance, and GPS support is only a 'nice to have' feature, and we are not supporting full configuration of GPS modules on power up, and are not planning to do so. Either get yourself a better quality GPS module that can persist settings, or else look into alternatives to Betaflight that have a focus on GPS and autonomous navigation, like iNav.