betaflight/betaflight #6632: Race setup optimization (feature request)

Ralfde, 2018-08-25:
My impression is, that betaflight does have a very good basic setting for freestyle, but a lot of users struggle to get a good race setup. It looks like the I factor is influenced by to many things at racing which makes it difficult to fly consitantly on a racetrack.
The racers also can not find any actual information  about a good race setting. Most videos describe good settings for freestyle.

As racing needs a complete different setup as freestyle, I suggest to add a button for preselection:
race or freestyle, which should adjust different basic settings, especially for all the iterm parameters in the background, which most normal users do not understand how to use them.
It looks like I is to much dependent on different things for racing like throttle changes and so on.
For constant flying on a racetrack the I-factor need to be more consistant and not discharged at throttle changes or at the first 50% of the stick travel for roll, nick and yaw. This will cause more propwash, but better control at flying the race track. Racers complain about this, that they can make I higher and higher, but it does not improve the handling on the racetrack, because it looks like the I-factor is influencend by to many things.
1.)
For the first step a dump of a good racer setup for betaflight 3.5 would help the most racers. But I think there should be also added a selection for freestyle and race, to help the normal users.
As betaflight does have different profiles, maybe Profile 2 could be used for racing and could be renamed.
f.e.
Profile 1 = freestyle
Profile 2 = race
Profile 3 = personal settings

2.)
Another thing I did not find is, that I could change the filter settings by radio channel.
After a crash and maybe using the turtle mode I would like to switch on all the notch filters, to make sure the quad flies also with unbalanced propellers.
So I miss a feature to switch on all notch filters by mode switch.

3.)
It would be also very good, if notch filters could be adjusted by a radio channel in the flight.

mikeller, 2018-08-26:
@Ralfde: You raise a couple of good points. One of the goals for Betaflight 4.0 is to rework how configuration is done, and to make it possible to load the default configuration for a target as part of the flashing process (i.e. not have it baked into the firmware binary any more). This will shift the onus of establishing good default values to the target maintainers, and also make the process to change defaults quicker. As part of this, target maintainers should definitely consider using the different profiles to define defaults for different flying styles.

Re your points 2.) and 3.), I think one of the things that needs to be done for 4.0 is to revisit the scope of profiles and rate profiles - as you say, things like filters should probably be moved into profiles or rate profiles, so they can be easily switched from the TX.

ctzsnooze, 2018-09-18:
How would you change the PID settings for a race quad?

closed 2018-10-25 by stale[bot]