betaflight/betaflight #6633: Preferred voltage for quad

PiManASM, 2018-08-25:
A lot of the time when we plug in, the battery voltage is detected at 25.2V/4.2V a cell, with 4.30 max cell voltage. Beta flight detects it as a near dead 7S, and flashes low battery for the entire race. It would be nice if the quad profile could include a standard cell count, as this is pretty constant for most quads.

timman2er, 2018-08-25:
@PiManASM you can fix this. Look in this thread of #4151 .

That being said, I see some merit to this. Some people however fly on multiple cell counts, which having an auto detection is nice for.

A command line force-setting cell count should probably be best, as I think it's a minority that really wants it.

mikeller, 2018-08-26:
Duplicate of #4151.

closed 2018-08-26 by mikeller

PiManASM, 2018-08-26:
The voltage appears accurately (4.2V/cell, 6S), and the max cell voltage was 4.3V/cell. It still displayed in the OSD as a dead 7S. The recommendations in that thread are already true of the quad. It seemed to be random as to whether or not the battery would display as full or not.