betaflight/betaflight #6634: Smart Audio Submenu empty

joeynuggetz, 2018-08-25:
Running BF 3.5.0 with TBS smart audio and anytime I change anything on the VTX via OSD the changes work as expected and are immediate. However, rebooting the FC and going back into the smart audio OSD menu will then show no options or an empty smart audio submenu. Is this a bug? Only way I found to clear this condition is to do something like set vtx_channel = 1 via CLI.

FC is betaFBV F3 with OSD/integrtated FRsky
Camera/VTX is AIO AKK Smart Audio 
Running AKK branch.

jflyper, 2018-08-26:
@joeynuggetz 

Are you saying you can not get the vtx online once it fails to get online after reboot, even with complete system power cycle?

joeynuggetz, 2018-08-26:
Well it broadcasts on same channel/band as usual and is functional after reboot it just affects the smart audio menu. Its just empty (nothing to select). Like I said, only way to trip it back to displaying the menu is to change channel via CLI. Perhaps other commands would do it to but that's just what I've been trying.

jflyper, 2018-08-26:
Full power cycle (FC and the VTX) does not bring the VTX online?

joeynuggetz, 2018-08-26:
The problem isn't with the VTX not coming online. Cam/VTX still functions fine after reboot of FC. Its just the menu in OSD is empty.

jflyper, 2018-08-26:
I'm referring to the state that the FC and the VTX talks to each other as online.

jflyper, 2018-08-26:
As displayed in the CMS menu.

joeynuggetz, 2018-08-26:
I see. Well I'm not sure how I could be able to confirm that. In some builds I recall seeing a statx menu but that's empty so I cannot tell if the FC and VTX are still talking to one another.

jflyper, 2018-08-26:
If the VTX is online, then CMS menu entries will be filled.

jflyper, 2018-08-26:
So, back to the question: "Does full system power cycle fix it or not?"

joeynuggetz, 2018-08-26:
Yes, and I did confirm that to be the case when I had the wiring incorrect the first time. The only thing I can confirm is that the VTX/cam function normally its just that the menu is affected. I've tried power cycling just the FC and only the cam/VTX and its the same unless I force a channel change via CLI or else the SA submenu will stay empty forever.

Full system power cycle doesn't fix it. Only fix I know of is set vtx_channel = x . Haven't tested other commands.

jflyper, 2018-08-26:
Hmm... Additional questions.

- What is the exact model of the AKK VTX?

- Have you had the particular VTX work flawlessly with earlier releases of Betaflight?

joeynuggetz, 2018-08-26:
ITs know as AKK AIO-SA

https://www.akktek.com/products/micro-aio-cameras/akk-aio-sa.html

I just got the camera and with the AKK implementation issues I've been reading about I'm not sure if that's still a "thing" with this model or not. I haven't found a version where it just works as intended but I'd be glad to try earlier versions if you want to maybe recommend a build and get back to you.

jflyper, 2018-08-26:
Can you still making changes and apply them from the new channel?

joeynuggetz, 2018-08-26:
I'm pretty sure but I will test that real quick.

joeynuggetz, 2018-08-26:
Did some more testing. Looks like I can change the band, power etc and switch to new channel on goggles and everything's ok and I can make further changes. What I did just notice is if I power cycle the VTX/cam while keeping the FC powered, the problem seems to go away and I was able to make changes even after a FC reboot. Does the VTX update the FC or the other way around? I'm gonna test some more to make sure this is 100% repeatable but for now it seems that functionality is maintained as long as I power off the VTX before the FC. However, I've been doing most of my testing with the USB connected and dont know why I didn't notice this before. What do you think?

joeynuggetz, 2018-08-26:
Looks like this is pretty repeatable. Just did a few more tests with channel changing and as long as I power off the VTX before rebooting the FC the SA submenu is there and I can make additional changes. Ran into a guy on the FB groups that had the same issue. Not sure if he was running the same vtx tho.

jflyper, 2018-08-26:
Can you find out the hardware model the guy on the FB is using?

Also, do you have any idea when this model was released? (How long has it been on the market, I mean.)

joeynuggetz, 2018-08-26:
Guy on favebook:

“I've had this on the TBS nano on Talon and HGLRC F4”

Regarding the release date of the akk vtx I’m using I’m guessing its pretty new as there were no reviews when i bought it. My google-fu shows most reviewers were posting vids about this product in Jan/Feb 2018

jflyper, 2018-08-26:
Nano? Tell him to come to the BF slack on #support channel. He needs a resistor if he doesn't have one already.

closed 2018-10-02 by stale[bot]