betaflight/betaflight #6640: BF3.5 about GPS return？

ykyandy, 2018-08-26:
This GPS return function begins to drop. Is the distance in meter? What is the reduction rate?
If I set the default 200, does it mean that it begins to drop 200 meters from me?
Finally, it landed 200 meters away from me.

set gps_rescue_descent_dist =[number] (default is 200)
**This is the distance at which your quad will start descending towards home.**
I think the installation manual only explains this. I don't quite understand the meaning of this sentence.

timman2er, 2018-08-26:
I think it indeed means that at that distance the quad will also descent.

callagga, 2018-09-02:
@ykyandy how did the landing go? Soft/hard?  When did it disarm exactly?  Guess I'm wondering if I test GPS rescue to the end including landing do I need to be ready to do some post repairs? 😀

ykyandy, 2018-09-20:
> @ykyandy how did the landing go? Soft/hard? When did it disarm exactly? Guess I'm wondering if I test GPS rescue to the end including landing do I need to be ready to do some post repairs?

I don't have time to fly the FPV now, but I've temporarily set it at 5 meters and I don't want it to start landing too far from me. Even a hard landing. I'll tell you when I go back to the test.

ykyandy, 2018-09-20:
> best is to watch some youtube videos.
> coders usually dont have time i guess to write manualls/wiki properly.
> 
> correct - distance at which your quad will start descending towards home.
> in my experience and test's this is just intended to get your quad out of the dead / no signal zone. i never let my quad land seeing what happened to others on youtube.
> i advise adding some serious LC filter on GPS power input to avoid ripples/mistakes in functionality.
> cheers.

YOUTUBE is not seen in China. T.T
Next time I'll test it myself. First find a larger lawn test.

closed 2018-10-27 by stale[bot]

yo8aiv, 2019-01-31:
any results ?..