betaflight/betaflight #6644: Design idea: Simpler radio connections for new pilots

davecraige, 2018-08-27:
What if Betaflight was a lot easier for new pilots when they connect their first new drone? Here is a mockup of the first screen of a binding wizard that could make it much simpler for new pilots.

I tried to set the size of the different radio brands correctly by drone usage.  Feel free to share any feedback you have.  Thank you 🙏

[image]

DieHertz, 2018-08-27:
And what exactly should happen when user clicks on one of them?
Binding is not a responsibility of Betaflight and out of it's control

wind0r, 2018-08-27:
I also dont understand how this should work. Some "magic" still needs to know whic UART is to use and also "FrSKY" / "Spektrum" arent any Protocol each of those companys has multiply protocols so there needs to be even more "magic".

Mockups are great and there is plenty of room to improve the user experience but without any technical informations or background those arent helping that much.

I think everyone already has had the idea to make the inital setup more easy without a technical idea this is not helping.

mikeller, 2018-08-27:
@davecraige:
- as @DieHertz says, 'binding' (as in binding an RX to a TX) is not something that can be initiated by the firmware, so this is out of scope. Did you mean 'configuring the firmware to use a certain type of RX' when you said 'binding'?
- this should be a feature request for the Betaflight configurator, not the firmware;
- your proportions are out. The correct order of popularity is more like (FrSky 55%, FlySky 15%, Spektrum 15%, Crossfire 5%);
- I can't wait to see your pull request to add this wizard to the Betaflight configurator. Thanks in advance!

timman2er, 2018-08-27:
I do think this'd be incredibly helpful for new users. I've got an idea of what should happen. Also, beautiful design!

@wind0r 
When you click on the brand, it could show a list of of transmitters and corresponding receivers. A config file that'll be really small contains the protocol for this type of receiver and other settings that we know from this info. Then, the user is prompted with providing info that we don't know, with an explaination for how they must get this info (e.g. which uart, they should look for naming on the FC).

@mikeller 
This is not OP's request, but once the above is done we could provide a finish screen along with default instruction for binding and setting up arm/flight modes along with explaination.

I think that should provide a full setup that makes it easier for users.

dribbe, 2018-08-28:
I think this is a great idea....
Could preset channel order, center, direction from center (reversing), and travel volume at some known/attainable Tx setting (like set tx to 125%, and all will be quite close).
All the Futaba and Spektrum based stuff is quite consisting/standardized as a rule. Many others are as well.   Personally, I am not as worried about the protocol/comm method, but it would be nice to have it all organized nearby on the app.

varkokonyi, 2018-08-28:
Horrible idea. Please forget about this as soon as possible

timman2er, 2018-08-28:
@varkokonyi would you mind explaining why?

mikeller, 2018-08-28:
@timman2er: Are you volunteering to implement and maintain this? Considering there are dozens of different RX models available across all brands, this will be a lot of work, for very little reward, since even if the 'wizard' had information about all of the RX models, without also having the corresponding information about the 100s of board designs and pinouts (and variants of boards using the same target), this information will be of very little use.

@dribbe: I think you are misunderstanding the original poster. The 'feature request' is titled 'simplified radio connections'. The problem with that is that the difficult part of the 'radio connection' is the physical wiring, and getting this right will require reading of the documentation for RX and board, and then correct application of the findings - this is not something that a wizard can help with.
What you are talking about are presets for ranges. This could probably be done in a wizard - but the digital protocols used today don't need any custom settings there any more - all commonly used protocols default to 1000 - 2000.

varkokonyi, 2018-08-28:
@timman2er Let's say you have a FrSky radio. Based on the receiver and the FW on it, it can be outputting SBUS, FPort, PPm or even good ol' PWM. And you still have to tell the FC which UART or pin it's connected to, along with th inversion on non-F4 boards. So it doesn't help at all

timman2er, 2018-08-28:
@varkokonyi agreed

@mikeller no I won't volunteer. I can't even get a normal PR out for a normal feature without having you guys hold my hand.

dribbe, 2018-08-28:
I probably changed the concept a bit from the OP, Sorry for that.

I disagree with you on the conclusion that they all default to 1000-2000.
examples:
SPM needs a couple of channels reversed in the tx, and needs the travel set to about 148% to get there.

FUT needs a couple of channels reversed, needs the centers (1520ms equiv) changed in the BF or in the Tx, the travel volume I still have not completely sorted... not sure if it can get there with default BF settings. Kind of a mess.
Also, the FUT in my first try (will try more).... when I set the travel for the lower part of throttle so it would ARM at center trim...  If I went to low trim, it would put the data OUT OF RANGE in BF... and it would basically 'lock-up' all Tx input... it seems the BF Rx input program is not well protected for inputs out of range.... it does not seem like it 'wraps' around, but it does basically seem to shutdown the entire input.

My point is that it is not completely easy or foolproof to use all the modern radios. I have not spent enough time with this to draw conclusions or point to real bugs, but those were my initial observations.

Just trying to help.

David

DieHertz, 2018-08-28:
>@mikeller no I won't volunteer. I can't even get a normal PR out for a normal feature without having you guys hold my hand.
Nice joke, except half of your PRs do not pass CI checks or don't even compile.

timman2er, 2018-08-28:
hahaha goosd sport @DieHertz

mikeller, 2018-08-28:
@dribbe: Ok, I did not take these into account. If there are standardised 'non-standard' ways to send channels over the digital protocols (that can and should be fixed with firmware only settings), then there will be some value in this. Still, calling it 'simplified radio connections' is misleading, as this is not really a question of connecting, but more of scaling once the connection is made.

dribbe, 2018-08-30:
Yeah... I don't have a crystal clear picture on the BF interactions with radios like Futaba (SPM is pretty easy to configure... but I may look at it again in the coming weeks... The out of range 'data lock' thing was a bit of a surprise for me.
If I get a more specific picture, I will advise. (I also want to check again to see if the Fut can do the 1000,1500,2000 thing with transmitter only settings... arm at center trim, and not 'lock-out' the data at low trim. The last pass I made at this I pretty quickly started making BF settings changes to see if it could be more safe/perfect.
I could imagine a wizard, or at last a ,settings script' to work in conjunction with 'known' transmitter settings to make it work safely with minimum settings.
FYI, in an ODD way, FRSky which is popular in the multi-rotor world did a 'partial emulation' of futaba s.bus, but they did not emulate FUT center, or reversing, or travel volume, so it makes it difficult to make the setup 'automatic'. It could really be two different choices for s.bus (FRSky or Fut).
Regards, David

closed 2018-10-06 by stale[bot]