betaflight/betaflight #6645: Support for failsafe_procedure change via transmitter switch (failsafe modes)

callagga, 2018-08-27:
Provide mode support in betaflight for the different failsafe_procedure modes, to allow a transmitter switch to be configured to change the failsafe_procedure modes (when unarmed I assume).  The modes being (currently): AUTO-LAND, DROP, GPS-RESCUE

Example usage:  One could then have failsafe_procedure set to GPS-RESCUE when flying outside in a medium/long range type of flight, then set back to DROP when changing and flying in an indoor location or under trees etc.

avr39-ripe, 2018-09-02:
I think failsafe_procedure should be OSD configurable so no corner cases when signal is weak and control packets can be corrupted can alter safety feature.

mikeller, 2018-09-02:
@avr39-ripe: OSD is a good suggestion for this, agreed.

callagga, 2018-09-02:
I was just after some way to do it I guess. I'll update the title to be less specific

avr39-ripe, 2018-09-03:
One more comment. If we implement failsafe_procedure configuration from OSD I think it is worth to somehow inform pilot about this parameter ARM time, so may be we should display something loke ARM - FS=DROP or ARM - FS=GPS. It is VERY important to know/remember what do you config for FS.

callagga, 2018-10-10:
another reason for this would be to immediately start flying when your GPS is cold (and will time minutes to get a position) - then when you're you/GPS is ready to await a fix, then turn it on..

mikeller, 2018-10-10:
@callagga: You'll still have to come back and land to enable GPS rescue - how else is the system going to know your home location / altitude...

callagga, 2018-10-10:
I was meaning (just like the situation I had today) rather than having to wait minutes to fly on the first battery (as the GPS needed to "warm up"/lockon), could just turn off the requirement for a GPS before betaflight would arm, then go fly.   Then some battery packs later you could then enable GPS to then use the GPS Rescue facility via OSD.

callagga, 2018-10-10:
PS.  I might be mixing up a few different things here:  BetaFlight Requirement to Arm before getting GPS lock (if you're not using GPS Rescue for failsafe should it let you arm?), configuration of a GPS rescue mode in the modes tab, and setting failsafe to use GPS Rescue.  But between all these my thinking was having the ability via OSD in between flights to have either:

a) Will use GPS Rescue in failsafe situation => needs lock
b) Won't use GPS Rescue in failsafe situation => doesn't need GPS working/lock

Some some way to switch between these two via OSD...

etracer65, 2019-01-28:
OSD support to change the failsafe settings was added in https://github.com/betaflight/betaflight/pull/7363

closed 2019-01-28 by etracer65