betaflight/betaflight #6649: automatically flip over (turtle mode) in self-level

timman2er, 2018-08-27:
In issue: #4652 a user described that in turtle mode the quad would automatically flip over after a crash if he switched to self level. This should make it way easier for users to flip over after a crash.

This is manually initiated (no additional modes/switch needed) with self level. In that thread it was speculated that it would only work if all motors were spinning/older version of BF installed, so in order to implement this I think we should make self-level work with only 2 motors spinning in reverse. That way, it shouldn't have to be added as an option but rather be.

Self-level might automatically pick the motors that are up, but in rare cases a prop might be stuck and can break the quad. The solution would be adding a limit for how long it's allowed to "try" before stopping (disable self-level until turtle_mode has been deactivated?).

timman2er, 2018-08-27:
Note to @Vidalcris , did this work in angle and horizon? Any additional suggestion?

Notifying you because you mentioned in that feature request you have used this.

etracer65, 2018-08-27:
I don't believe this could be implemented safely or in a way that is unlikely to damage motors and ESC's.  As you mentioned, there's no way to know if the props are jammed.  Just having the flight controller apply throttle based on "which side is more up" is likely to cause lots of damage.  Additionally having the throttle out of the control of the pilot is not considered safe.  Lastly there's no way for the flight controller to know how much throttle to apply.  Different props/motor combinations make different amounts of reverse thrust.  So in this case the flight controller would have to apply a small amount of power and increase if nothing is happening.  Unfortunately we can't differentiate that from the reason nothing is happening is because the props are jammed.  So the flight controller applies more and more power and smokes the motors.

timman2er, 2018-08-27:
@etracer65 could the throttle not be in the pilot's control? That way, pitch and roll will be automatically done with self-level, and throttle by the user. That should solve this problem.

etracer65, 2018-08-27:
The problem is that in turtle mode the throttle does nothing.  The amount of power applied is based on how far the pilot moves the roll/pitch stick in that direction.  So if the pilot already has to move the stick that direction to control throttle then where's the "automatic" part?