betaflight/betaflight #6652: Configuration lost (reset to default) on battery very low, under 2.5 volt per cell.

vinz486, 2018-08-27:
Hi,
I don't know if this is an hardware fault. After a flight I gone over after battery alarm. Landed far away from me, quad left turned on until the 3S battery gone very low in voltage, under 2.5 volt per cell.

When I put a fresh lipo on quad I noticed one motor spinning and was unable to arm. When connected to BF configurator I found the entire configuration reset: everything was to default.

This is normal?

Thank you.

Betaflight: 3.5.0
FC: FuryF4OSD (Diatone board)
Quad: Diatone M2.506 X
Lipo: 3S GNB

timman2er, 2018-08-27:
Have you had this issue multiple times? Does setting min_voltage to under the voltage you run down to fix this?

jflyper, 2018-08-28:
No, it is not normal.

It is rare,  but there are reports of config resetting from time to time.

Is is possible if config storage was somehow corrupted, and it is reset to default upon next start up. The corruption may occur if the firmware was actively writing to the config storage when the board is reset, in your case, by BOR (Brown-out Reset). Writes to config storage only occur at very limited occasions, but stick commands are one of them. So, if you were fiddling with sticks randomly when the supply voltage went down, the corruption is possible.

One more thing to evaluate is how low the supply voltage should be to trigger BOR. My guess is less than 5V, in 3S case, 1.67V or less per cell.

vinz486, 2018-08-28:
Yes, voltage was low as you reported.

But it's not clear for me what is trying to write to storage: I actively disabled blackbox loggiing and I was thinking that anything should write data. Why stick movements write data? And where?

Thank you!

jflyper, 2018-08-28:
Stick commands will select active profiles, calibrate acc/gyro and write results, etc.

vinz486, 2018-08-28:
Oh.. right.

Thank you, I learned something today (cit. @joshuabardwell ).

Found some documentation https://www.embedded.com/electronics-blogs/break-points/4430317/Brown-Out-Reset

closed 2018-08-28 by vinz486