betaflight/betaflight #6653: Runtime GPS configuration when external GPS is used (via MSP)

awolf78, 2018-08-27:
I was hoping with https://github.com/betaflight/betaflight/commit/83fa3f31c67e8a2e0304905e868b14eed26e79d4 I could use GPS Rescue by pushing GPS data from ImpulseOSD. Unfortunately ```static void validateAndFixConfig(void)``` in ```config.c``` kills the GPS (rescue) related configuration if no serial port for GPS is configured :(

```
if (
#if defined(USE_GPS)
        !findSerialPortConfig(FUNCTION_GPS) &&
#endif
        true) {
        featureClear(FEATURE_GPS);
    }
```

and then a bit later

```
if (
        featureConfigured(FEATURE_3D) || !featureConfigured(FEATURE_GPS)
#if !defined(USE_GPS) || !defined(USE_GPS_RESCUE)
        || true
#endif
        ) {
        if (failsafeConfig()->failsafe_procedure == FAILSAFE_PROCEDURE_GPS_RESCUE) {
            failsafeConfigMutable()->failsafe_procedure = FAILSAFE_PROCEDURE_DROP_IT;
        }

        if (isModeActivationConditionPresent(BOXGPSRESCUE)) {
            removeModeActivationCondition(BOXGPSRESCUE);
        }
    }
```

I like the self-healing configuration approach, so I am looking for some advice from the main devs how I should approach fixing this.
I could:

- Remove the self healing in ```config.c```.
- Delay the GPS related self healing in ```config.c```. Problem is when you only connect USB then it won't get GPS data.
- Add another configuration entry that makes BF aware that there will be GPS available via MSP

Maybe you have a better idea, but I would be happy to implement anything that fixes this.

mikeller, 2018-08-28:
@awolf78: Lol, talk about stuff that comes back to bite you. I think the self-healing is an important aspect of how we do configuration in Betaflight. At the moment we are bad at providing feedback from the firmware to configurator on what features are supported by a particular target, and with 130+ targets it's absolutely impossible to maintain a feature matrix on the configurator side (plus this would not cover custom builds, which we want to encourage). In this situation, self-healing takes on the role of giving the user feedback when a feature (or a combination of features) is not supported on a particular target.

As to your problem, I think the best possible option is to add `GPS_MSP` to `gpsProvider_e` in `io/gps.h`, and change the code to not expect / initialise an UART if this is selected, and expect the GPS information to be injected via MSP in this case.

The reason for injecting at this point is that I would imagine that users might want to do other things with the GPS information they are getting besides GPS rescue, like logging it or displaying it in the OSD.

awolf78, 2018-08-28:
Thanks for the suggestions, created PR here: https://github.com/betaflight/betaflight/pull/6662

closed 2018-09-27 by awolf78