betaflight/betaflight #6654: Quad accelerometer severely mis-aligned after throttle punches

VoicOfReason, 2018-08-27:
I've been having this problem since last year and it has not gone away despite updating to the latest Betaflight over time. I have a Karearea Talon with 2307-2722 motors, running on 4S. I'm using 35A ESC 4-in-01 with Omnibus F7 V2. 

Here's the problem: when I power on the quad, in STABILIZED MODE, it flies level, and I can fly around as much as I want with no issues. But as soon as I punch the throttle, it instantly throws off the leveling and now it flies severely crooked, until I land and re-calibrate level. Granted, I don't fly FPV in stabilized mode, but this is a problem. 

Can Betaflight address this issue?

timman2er, 2018-08-27:
This sounds incredibly strange, as those variables should not change on throttle. 
How much is it thrown off? Could it be due to hardware or vibrations coming in?

etracer65, 2018-08-27:
This is usually caused by excessive vibrations from the motors/props/frame.  The accelerometer is quite sensitive to vibrations as they are instances of acceleration as far as it's concerned.  Work on reducing the vibrations (motor bearings, prop balancing, soft mount flight controller, etc.).

VoicOfReason, 2018-08-28:
It does the same thIng on different props. All motors are soft mounted and
rubber standoff’s used on FC. No visible vibrations at all in acro mode.

jflyper, 2018-08-28:
@VoicOfReason A blackbox log with the phenomenon is required for further discussion.

VoicOfReason, 2018-08-28:
Will do.

etracer65, 2018-08-28:
Please also include a `diff` of your settings as requested in the issue template.

VoicOfReason, 2018-09-27:
I'll try to get a blackbox uploaded in next few days.

DrBonesLCD, 2018-09-29:
I have this happening to two separate and different quads after I updated to BeF 3.5.1 - a Chameleon 220 frame with a Cicada AIO 4-1 30 ESC Omnibus F4 FC and with the HGLRC Omnibus F4 stack on a 145 frame. I put BeF 3.3 back on and voila - no more of the mis-aligning after throttle punch.

closed 2018-11-05 by stale[bot]