betaflight/betaflight #6657: KISS FC V2 "No DFU capable USB device available"

S-T-A, 2018-08-28:
I had a hard time to get the kiss FC to go into DFU Mode to start a fresh install of the betaflight_3.5.1_KISSFCV2F7-3-5609760df.hex
in betaflight it didn't work so i tried installing it from the KIss GUI 2.0.6 and it worked kinda i wasn't able to use the kiss GUI anymore; on betaflight standalone configurator 10.4.0 it opened up and worked! when i went to save after trying to configure the ports. i got locked out the FC is not readable can't set it to DFU I tried everything on mac and windows 10. I have never had a problem like this is it just a kiss thing? now i have 2 KISS v2 F7 boards that are both locked out! do i have to reset the clock???

mikeller, 2018-08-28:
@S-T-A: Sorry to hear, but for what I know your flight controllers are bricked irrevocably unless KISS can supply you with a fixing utility, or replace your boards:
- the KISSFCV2F7 uses a custom boot loader, and therefore cannot be flashed with the Betaflight configurator, but only with a tool provided by KISS;
- this is a known limitation, and is the reason Betaflight never released a KISSFCV2F7 firmware. You force-flashed an unstable development build, so I do not know if this is covered by the warranty from KISS;
- @ronlix might know more about how this can be fixed (if at all).

S-T-A, 2018-08-28:
Any help would be much appreciated this just sucks spending $150+ for kiss if @ronlix could help i would really appreciate anyones help at this moment...

mikeller, 2018-08-28:
I suspect that in your situation the only ones potentially being able to help are KISS / @ronlix. If the custom boot loader on your boards has been overwritten by flashing it with the Betaflight configurator, then they will have to provide a backup of it (and a way to install it) if your boards are to ever work like normal KISSFCV2F7 again...

closed 2018-08-28 by S-T-A

DieHertz, 2018-08-28:
KISS FC V2 F7 has no DFU, it uses a CP21xx USB to serial chip for design considerations

ronlix, 2018-08-28:
Hi,

yes we have our own bootloader there to make not have the windows DFU + readoutprotection problems.
as far as i know this bootloader is not brickable unless you connect to it over SWD or Jtag.
to enter the bootloader (no matter what FW is flashed) press the reset button once while its connected to the PC (USB). then the blue LED should filicker fast.
if you now connect to the KISS GUI (while the LED is flashing) it forwards instandly to teh flasher tab where you can falsh KISS or BF firmware.

to reset the eeprom teh simplest way is to flash once the KISS FC FW to reset the BF eeprom and BF FW to reset the KISS eeprom.

regards

Felix

mikeller, 2018-08-28:
@ronlix: Thanks for helping here!

@S-T-A: Can you please let us know if this works for you?

S-T-A, 2018-08-29:
YOU GUYS ROCK I WOULD KISS YOU IF I COULD!!! :)

mikeller, 2018-08-29:
@S-T-A: I take it that means it worked? :laughing:

Adisuciu90, 2018-12-13:
can we get an hex file for KISSFCV2F7?

mikeller, 2018-12-13:
@Adisuciu90: As the Betaflight configurator can not be used to flash Betaflight firmware onto the KISSFCV2F7 due to a custom boot loader being used, Betaflight hex files are provided by KISS. You can ask @ronlix to supply what you need.

Adisuciu90, 2018-12-13:
ok,thank you!

PalmTreess, 2019-12-12:
Does anyone have a link to the Betaflight hex files? I looked on KISS's website but couldn't find them.

mikeller, 2019-12-12:
@PalmTreess: We do not publish them as installing them with Betaflight configurator could potentially brick the flight controller. You'll have to contact KISS to ask them where to get Betaflight for the KISSFCV7F&7 and how to install it.

PalmTreess, 2019-12-12:
I sent a message over to @ronlix we will see. Seems silly that everyone would have to contact him though.

mikeller, 2019-12-12:
> I sent a message over to @ronlix we will see. Seems silly that everyone would have to contact him though.

Yeah, they should really publish the firmware on their website.

whig0, 2020-07-21:
> I sent a message over to @ronlix we will see. Seems silly that everyone would have to contact him though.

Did you get any answer about this? I'm interested as well :-)

Adisuciu90, 2020-07-21:
Nope,sorry

Trimis de la Yahoo Mail. Obțineți aplicația 

    Pe marți,