betaflight/betaflight #6665: gcc compile option -fno-common

robert-b, 2018-08-29:
bad coding can happen - let the compiler complain abvout it.
add -fno-common as a gcc option to the makefile

-fno-common
In C code, this option controls the placement of global variables defined without an initializer, known as tentative definitions in the C standard. Tentative definitions are distinct from declarations of a variable with the extern keyword, which do not allocate storage.

Unix C compilers have traditionally allocated storage for uninitialized global variables in a common block. This allows the linker to resolve all tentative definitions of the same variable in different compilation units to the same object, or to a non-tentative definition. This is the behavior specified by -fcommon, and is the default for GCC on most targets. On the other hand, this behavior is not required by ISO C, and on some targets may carry a speed or code size penalty on variable references.

The -fno-common option specifies that the compiler should instead place uninitialized global variables in the data section of the object file. This inhibits the merging of tentative definitions by the linker so you get a multiple-definition error if the same variable is defined in more than one compilation unit. Compiling with -fno-common is useful on targets for which it provides better performance, or if you wish to verify that the program will work on other systems that always treat uninitialized variable definitions this way.

never, never, never declare variables in a header file!

digitalentity, 2018-08-29:
See https://github.com/iNavFlight/inav/pull/3823 - enabling `-fno-common` revealed 3 cases of multiple declarations of the same variable. @mikeller FYI

DieHertz, 2018-08-29:
Agreed on making it default, ODR should not be violated.

mikeller, 2018-08-30:
Yeah, this seems to be a good step. Do you want to open a pull request for this change, @robert-b?

mikeller, 2018-08-30:
@robert-b: I couldn't resist checking what would be broken myself, sorry...

PR #6668 (2018-09-08): Added '-fno-common' build option, fixed resulting problems.
Fixes #6665.
closed 2018-09-08 by mikeller