betaflight/betaflight #6666: dump all  end in the middle

robschii, 2018-08-29:
when i want to dump all in cli comand the  dump ends with this screen....
[image]

robschii, 2018-08-29:
hglrc flame f4

etracer65, 2018-08-29:
Do you think you could at least tell us which version of Betaflight you're using and provide the requested `diff` or your settings?  Otherwise we're just guessing and likely won't be able to help.

This type of problem is usually caused by a corrupt configuration or upgrading versions without doing a full chip erase.

atshort, 2018-08-30:
I am also having this problem.

XJB-145 3" quad with Omnibus F4 (OBF4) controller. The stack is called "HGLRC F428 TX20 V2". My configuration is:

Betaflight 3.5.0, configurator 10.4.0
R-XSR SBUS on RX1
Tramp on TX3
RunCam Micro Swift v3 (the UART version) on TX6/RX6
Soft serial enabled on PPM pin, used for SmartPort telemetry.

When I do "diff" in the CLI the UI freezes part way through. In addition the entire FC appears to be frozen, it becomes unresponsive to any RC commands. If I disable telemetry (but have soft serial configured still) the problem does not occur. On the bench my CPU is about 8% without soft serial and 12% with soft serial.

Attached is a diff, without telemetry obviously since I can't get a full listing with that enabled.

Also, I have two supposedly identical quads, configured identically, and this problem occurs reliably on one and I can't reproduce it on the other.

The quad which shows the problem has been flying reliably for months and I've done a CLI dump many times. The problem only started when I added Smart Port on soft serial. I've done a clean wipe/install a couple of times and that didn't help.

Is there anything I can do to help narrow down the problem?

[XJB-PY config 082918-4.txt](https://github.com/betaflight/betaflight/files/2334252/XJB-PY.config.082918-4.txt)

atshort, 2018-08-30:
Addendum to above. I did a lot more tests and found the problem sometimes occurs with telemetry turned off, and occasionally with soft serial off too. If I turn off telemetry, soft serial and LED support I haven't been able to reproduce it so far.

etracer65, 2018-08-30:
@atshort I tested with your config (and enabled telemetry) on an OMNIBUSF4 target (real Airbot version) and couldn't replicate.  Does it hang during a `dump` command as well or only during a `diff`.  In either case does it always hang at the same spot in the output?  If so, can you provide the outputs generated?  Seeing the items before the problem may help point to the issue.  I don't have any HGLRC flight controllers so hopefully it's not something specific to their boards.

atshort, 2018-08-30:
So far it has frozen at a different place each time. When diff is hanging dump is hanging too.

I am quite confused at the moment. I left it powered down while I ate dinner then came back and everything worked with telemetry, LED, and soft serial on. I ran dump and diff about 40 times without a problem. Then I toggled feature LED_STRIP off and on and immediately diff and dump started freezing every time. So that doesn't make sense to me.

I wondered if there was a temperature sensitivity issue with the board, but some tests with a fan and a hot air gun suggest not.

I need to think about this some more. It isn't clear whether this is the board/bf, the configurator, my computer, or what. The fact that I have two identical quads and can't reproduce the problem on one of them it is a mystery too. I reported this here because someone else already had this thread going and the problem sounded the same. But I'm definitely not ready to suggest this is a Betaflight software issue.

I'm on a Mac. Do you happen to know if there a way to get to the CLI without running the GUI configurator? That would help eliminate one component from the test. Going to try another computer too.

mikeller, 2018-08-30:
@atshort: Thanks for testing - your description hints at the feature configuration as a potential source of this problem, which makes it likely that it's caused by #6254.

How did you toggle the feature (UI / CLI), and did you save the change, or abort it?

etracer65, 2018-08-30:
Well I tried the above procedure and still could not replicate. But possibly this leads to some issue with the way features are managed as referenced by @mikeller.

mikeller, 2018-08-30:
@etracer65: On second thought #6254 can't be the culprit - it was merged after 3.5.0. Hard to imagine how this could be related to this problem, but HGLRC is known for bad quality hardware, there are other common problems with these boards because of it.

nerilorenzo, 2018-08-30:
I had the same behaviour with kakute f7 betaflight 3.5 akk version of 15-8-2018
I don't know if help you the information that I have done a lot of setup for flight mode and osd and probably the output of the diff command is too long.
I tried also with diff all and I had the same behavior.

flint723, 2018-08-31:
I had the problem once, on matekf405 mini.  Reflashed to newer master (BF 4.0 1081 to 1082) and have not had the issue again.

Hydroculture, 2018-09-03:
Same dump problem happens to me, stops in the middle of the vtx section of dump/diff.

The hardware used is HGLRC xjbf440, rxsr/sbus

I am running firmware released on: Sep 3 2018 13:23:17

I did not have this problem with release  #1084 Date: 27-08-2018 12:13, same hardware

Quad is still flying extraordinary well with bf4.0.0! 
Thank you for your brains

best 
Holger Landwehr

etracer65, 2018-09-03:
Since the problem appears to be limited to HGLRC boards and none of us have these boards to test, somebody that can replicate the problem is going to have to see if they can narrow down to which build the problem first appears (or the last build that the problem doesn't exist).  Also while testing be careful to replicate the tests as it appears to be somewhat intermittent.

Normally this is done with a process called "bisecting".  If you know a recent build that doesn't work, then that's your upper boundary.  Pick an older build and try that.  If it also doesn't work then that's your new upper boundary.  If it does work, then that's your new lower boundary.  Pick a build in the middle of your upper/lower and test.  If it works then it's the new lower boundary; if not it's the new upper boundary.  Repeat until you hone in on the problematic build.

Once identified we then need other people with the same problem to confirm.  Once the build is identified then we can look at the changes from the previous and possibly see a problem.

Otherwise it's extremely difficult for us to debug without access to the hardware.

Hydroculture, 2018-09-03:
@etracer65, thank you.
 Ill search for the boundaries this evening

best
Holger

Hydroculture, 2018-09-03:
Grrrr! the 'beloved' error doesnt occur anymore

Started from 
#1084 - ok
#1091 - ok
#1096 - ok
#1098 - ok
#1099 - ok
Procedure was:
Erased
flashed
pasted my dump
changed osd font
changed logo
disconnected
reconneted
             
then "dump all". No Problems. 

Doubletested on a brand new hglrc xjbf440 and the copter that constantly showed the dump-all-error.

Very sorry that I could not help. Still happy its working atm. Magic.

best
Holger

etracer65, 2018-09-03:
@Hydroculture Thanks for trying.

A question for others that can replicate the problem: Have you tested immediately after a "full chip erase" flash or after doing "defaults" in the CLI?  If it works with clean defaults than it might be related to some particular setting after you restored your configuration.  May have to step through your `diff` that you're restoring to see if the problem reoccurs after a particular setting.

atshort, 2018-09-03:
I described my procedure for reproducing the problem [here](https://github.com/betaflight/betaflight/issues/6666#issuecomment-417185054) and used a full chip erase. I've had it happen in other ways too, but the procedure I described required changing just a single setting in the configurator GUI.

I have a total of four (identical) HGLRC flight controllers that demonstrate this problem. In my experience so far I haven't seen anything that impacts flying, this freezing is a bench issue when using the configurator.

Note there are two other flight controllers mentioned in this thread as possibly having the problem - a Kakute F7 and Matekf405.

I'm fairly sure this problem appeared after I moved from 3.4.0 to 3.5.0, but I haven't done a rigorous test of that. I'll try to get a definitive answer to that question.

AndersHoglund, 2018-09-04:
This is not only a HGLRC problem. I see this on SpeedyBeeF4. It seems you have to change some setting in the configuration tab to see the problem. But only with a base diff pasted from a file. I.e. the procedure outlined by @atshort. 

I also noticed the latest 4.0 builds does NOT show this problem. I did a bisect to find out in what 4.0 build and it seems build 1094 fixes this while 1093 and older has this problem.

Next a bisect to find out what build in 3.5 that introduces the problem, and I find that build 1001 and older are OK while build 1002 fails.

Build 1002 integrates a MSP change: "Remove MSP BOX BME entries for baro, gps home" PR #6318 that seems to cause this issue.
Build 1094 integrates PR #6679, but how this PR can fix this issue makes no/little sense to me. It would have been more logical if build 1095 made a difference as it contains MSP changes. But this is what I have found out so far, it would probably be a good idea if someone else could please confirm those two bisect results.

AndersHoglund, 2018-09-04:
To be complete, this is the base config diff I used: 
[betaflight_diff_20180826.txt](https://github.com/betaflight/betaflight/files/2348316/betaflight_diff_20180826.txt)

To force the failure, paste the diff from above file, reconnect and open the "configure" tab and turn LED_STRIP off. Save and reboot. Open cli and do a "diff" or "diff all".
The "diff" and "diff all" commands never fails directly after pasting the config, you have to make some change in the "configuration" tab to make it fail.

mikeller, 2018-09-04:
@AndersHoglund: Interesting, thanks for bisecting. I tried to reproduce the problem with your diff / instructions on an OMNIBUSF4, but the `diff all` at the end worked without hanging.
Tried again with a SPEEDYBEEF4, and this hangs reproducibly (side note: it hangs in different places in the `diff all`).
Looks like this is also target dependent.
Testing build 1094 with SPEEDYBEEF4 works fine, like you say.

mikeller, 2018-09-04:
@etracer65: Could it be that it's only changing features that is causing this, or does changing other parameters have the same effect?

etracer65, 2018-09-04:
@mikeller It looks like simply "Saving" on the configuration tab causes the problem.  No need to change anything.  Doesn't appear to be feature change related.

etracer65, 2018-09-04:
Also saving on other tabs doesn't cause the problem - even in cases of "Save and Reboot" like the ports tab or blackbox settings.  Just seems to be related to the configuration tab.  Likely something in the byte sequence is out of order (or missing) and the config is getting offset/invalid data.

Also tested on current 3.5 maintenance head and it does still have the problem.

btandrei, 2018-09-04:
if you set Smoothing Type to Interpolation also fixes the problem.

etracer65, 2018-09-04:
Ok, I pretty sure I found the problem.  It's a disconnect between the firmware and the Configurator where the USB HID mode flag is being sent/received by the firmware but it's not supported in the Configurator yet.

Can those that can repeat the problem try one the appropriate hex below and report back?  Let me know if another target is needed.

[betaflight_3.5.1_OMNIBUSF4.hex.zip](https://github.com/betaflight/betaflight/files/2349819/betaflight_3.5.1_OMNIBUSF4.hex.zip)
[betaflight_3.5.1_SPEEDYBEEF4.hex.zip](https://github.com/betaflight/betaflight/files/2349820/betaflight_3.5.1_SPEEDYBEEF4.hex.zip)

AndersHoglund, 2018-09-04:
@etracer65 You hex file for SBF4 worked just fine. 
```
# version
# Betaflight  SPEEDYBEEF4 (SBF4) 3.5.0 Aug 14 2018  035721 (1d710afd7) MSP API 1.40

```

etracer65, 2018-09-04:
Submitted fix https://github.com/betaflight/betaflight/pull/6696

PR #6696 (2018-09-04): Fix USB HID parameter support in MSP_SET_RX_CONFIG msp message
Fixes #6666 

Workaround to make the firmware tolerant of missing Configurator support in versions 10.4.1 or less. Will still work properly when Configurator support is added.
closed 2018-09-04 by mikeller