betaflight/betaflight #6667: FPVM_BETAFLIGHTF7 BLHeli ESC pass-through won't work since 3.4.0

ButschFPV, 2018-08-29:
After flashing 3.5.0 I tried configuring my FVT_LittleBee 30A ESCs (BLHeli 14.9) via pass-through.
There are no ESCs showing up in BLHeliSuite and BLHeli-Configurator.
I switched back to 3.3.3 and it worked just fine.
With 3.4.0 and 3.4.1 sometimes 1-2 ESCs are shown but writing config fails.
Settings are all left on default.

jflyper, 2018-08-30:
Tried with Omnibus F7 (V1) which is not identical, but functionally equivalent board, flashed with FPVM_BETAFLIGHTF7 target. ESCs are Airbot Typhoon Race BLHeli_S 4-in-1, flashed with BLHeli_S 16.7.

3.4.0 and 3.5.0 both work with BLHeli-Configurator.

This test result is not conclusive, but I suspect configuration and/or hardware (including peripheral firmware) problem.

3.4.0
[image]

3.5.0
[image]

ButschFPV, 2018-08-30:
I desoldered everything to exclude issues from peripheral.
With only one Motor and ESC hooked up to M1 the error still occurs.
I attached my logic-analyzer for troubleshooting and after that for some reason the ESC shows up in BLHeliSuite.
Long story short, I solderd a 10k pull-up from M1 to 3V3 and now the pass-through works with M1 and 3.5.0 just fine.
I don't think I fried the internal pull-up because with 3.3.3 it works.
Strange. Any Idea?

jflyper, 2018-08-31:
@ButschFPV 

> I don't think I fried the internal pull-up because with 3.3.3 it works.
> Strange. Any Idea?

Weird. The `IOConfig` call that governs the signal condition hasn't been changed for a year.

How about this test if you haven't installed rest of the pull-ups? Instead of the pull-up or logic analyzer probe, what if you touch the ESC signal pad while connecting?