betaflight/betaflight #6674: What happend with BARO-Mode in 3.5.0? Matek F405

MR444, 2018-08-31:
I just flashed my Matek F405 to v. 3.5.0 but I cannot find the BARO mode anymore. Was it removed? I thougt only F3 boards do not have this function anymore. 
GITREF 10710AF
Target MatekF405

fujin, 2018-08-31:
Baro mode was removed when GPS Rescue was introduced.

closed 2018-08-31 by fujin

MR444, 2018-08-31:
But I do not have GPS receiver installed. Is there no mode for altitude holding anymore?

fujin, 2018-08-31:
No, there is not.

MR444, 2018-09-01:
> What was in 3.4 was a switch for BARO mode, but that switch did nothing at all - when the broken BARO mode was removed it was forgotten to remove the switch as well...

Okay I did not notice that. 3.4 was the only version where I did not use baro mode. In the versions before I used it.