betaflight/betaflight #6675: Baro mode - want back :)

MR444, 2018-08-31:
When I flashed v3.5.0, I saw that baro mode was removed. I do not know if anyone used it but I would like to have it back. 

Is there space left for this flight mode? In v3.4.0 there was baro mode AND GPS rescue mode.

andrejpodzimek, 2018-08-31:
I use it (or at least try to, because it has a number of glitches) and would like to keep using it (if it has really been removed).

Also, the default ALT PID and VEL PID values in Betaflight shoud be adjusted so that baro hold can work without crazy throttle oscillations out-of-the-box:

* ALT PID 50, 0, 0
* VEL PID 50, 5, 30

It's very surprising that VEL PID can influence baro hold stability. Further context is in #3823.

I also have a feature request #3824 open to fix / properly define and document the meaning of the throttle channel in baro mode.

andrejpodzimek, 2018-09-01:
@mikeller Oscillations were only a matter of the two PID values mentioned above. The defaults were wrong, that’s all.

The one and only problem was with the meaning and function of throttle in baro hold mode — that was not properly defined.

etracer65, 2018-09-01:
If you really want to use altitude hold (baro mode) on Betaflight, perhaps you should ask yourself why you’re not using iNav instead. What value do perceive having altitude hold adds to Betaflight?

mikeller, 2018-09-01:
@andrejpodzimek: The fact that certain PID settings can make it work for one particular setup does not mean that it can be made to work for any other setups.

The problem is that the entire altitude estimation code was dubious and had been for years before it was dropped. Since nobody had been stepping up to put in the effort to make it work (and from the look of it, iNav is putting in a lot of effort to make these things work), the decision was made to drop this code. If somebody was to fix it and make it work then re-adding it would certainly be considered on targets that have enough flash space.

andrejpodzimek, 2018-09-01:
@etracer65  I’ve said this a gazillion times under different bugs: Because iNav doesn’t have DShot. That’s a showstopper. As for value, I don’t fly FPV and to fly to the edge of the line of sight, which I do quite often, altitude hold is a very important feature.

@mikeller The same can be said about the stock gyro PID settings — there might be setups where they won’t work well. Yet they do work on most builds without tuning. It would be great to have such conservative working defaults for baro hold, that’s all. An alternative option would be to implement DShot in iNav, but that would be certainly orders of magnitude more complicated than baro in Betaflight.

MR444, 2018-09-01:
> As for value, I don’t fly FPV and to fly to the edge of the line of sight, which I do quite often, altitude hold is a very important feature.
Even with FPV, baro mode can be helpful. At 25mW (only way to fly legal in many countries) you can loose FPV signal easily. With baro mode, I could bring it back without any GPS function. Baro mode was something like panic button for me.
But andrejpodzimek is right, it was hard to find working PID ratings without oscillations. In the new versions I wish default ratings similar to andrejpodzimek's.

mikeller, 2018-09-02:
@andrejpodzimek: I think you are ignoring the fact that the PID controller has been designed and tuned with a focus on making it work well for most setups. The altitude hold code was just a patchwork that seemed to be not unusable for some setups with some PID values. The difference there is 'a lot of work' vs. 'no work at all', for years.
As I said before, if somebody wants to put in the work to implement a proper altitude hold in Betaflight, or probably even better, port iNav's altitude hold code into Betaflight, we are not opposed. But your assumption that this will be less work than porting DShot to iNav is simply wrong.

andrejpodzimek, 2018-09-02:
The way I see it, baro is working fine on the hardware support level and the only missing bit is a better feedback control algorithm for it and also improvements to the baro mode, such as a better way of interpreting the throttle channel input in that mode.

DShot looks like a completely different story. I still remember quite vividly the all-nighters spent on Betaflight builds where 2 motors out of 6 or 1 out of 4 wouldn’t work until someone gave me lots of advice on how to repurpose some outputs in order to get the hardware timers needed for DShot. (The removal of TX DMA saved the world for me, because without it even old F3s can do DShot on all motors.) That said, if iNav has been developed and maintained without DShot in mind, getting it to work (without breaking lots of other things) could be a nightmare. (If it was easy, someone would have already done it. :grimacing:)

As for porting or re-implementing altitude hold, that could be easier to do if the necessary scaffolding (e.g. the baro hold mode) was still in place. The removal of (partially) broken features (compass, baro, GPS hold) basically drops them on the floor indefinitely — if I decided to (e.g.) spend a weekend looking into this, the need to get a working branch first (by digging some stuff out of history) would be quite a deterrant for me. Coding I do for a living differs way too much from both MCUs and the JS configurator stuff and I don’t feel like (e.g.) taking vacation and making learning investments there when I see flight modes being actively removed from release to release.

etracer65, 2018-09-02:
@andrejpodzimek What's the deal breaker with iNav not having DSHOT support?  iNav does support multishot so protocol speed isn't an issue.  Also BLHeli_S and BLHeli32 support multishot natively as well.  The only things given up are turtle mode and possibly DSHOT beacon - both of which seem not applicable to your description of LOS flight.

digitalentity, 2018-09-02:
@andrejpodzimek @etracer65 INAV will support DSHOT in version 2.1 - while I don't find it very useful, demand for it is too high to ignore 😄

andrejpodzimek, 2018-09-02:
@etracer65  I don’t care about protocol speed; there’s nothing that a human could notice about it anyway, despite all the hype. What I do care about is, besides the beauty and calibration-free precision of a digital protocol (although Proshot is in fact a much more “beautiful” idea), **ESC telemetry**. Telemetry is the *killer feature* of DShot. I always record my OSD signal (which is why I always install OSD despite not using FPV at all) and it’s just fancy to see the RPM and other stuff from the ESCs.

@digitalentity Now that’s purely awesome! I’m looking forward to trying it out. I guess that my equipment won’t work there either, because all of my 3" builds packed with GPS+compass+baro have severe problems with interference on the first two items — the compass just makes a slow 360° turn in 2 minutes when you let it sit on the bench and watch the numbers, perhaps due to the proximity of other parts, and the GPS **never** works unless I switch the RunCam Split off, at which point GPS does work, but then I have no video at all, analog or digital. All of this makes me tremendously curious about iNav — I guess my hardware issues will still be issues, but at least the baro mode could be better there! :smile:

andrejpodzimek, 2018-09-02:
[image]

This crowd has just gathered on my table, screaming *iNav!!!* *iNav!!!* — they are all curious about it.

Equipment (left to right): baro+mag, all 3, all 3, all 3, all 3, baro
MCU (left to right): F3, F4, F7, F7, F7, F3

Well, if I could do at least a baro hold, that would be nice. They are all tired of my triple-flips and would like to try some calm flights as well. For the GPS, only the big one has a good fix. On the others, experiments with EMI shielding tapes are pending. I still don’t understand why GPSs work so well in cell phones, but have so much trouble in a small drone.

But anyway, all of them want at least a baro mode to start with — so they say.

mikeller, 2018-09-02:
@andrejpodzimek:

> As for porting or re-implementing altitude hold, that could be easier to do if the necessary scaffolding (e.g. the baro hold mode) was still in place. The removal of (partially) broken features (compass, baro, GPS hold) basically drops them on the floor indefinitely[...]

That is true. But on the other hand, the broken code was in there for years, and nobody ever cared enough to fix it. So when GPS rescue was done, and there was a choice of either making the good new code work around the old broken code, or removing the code that nobody had cared about for years, the decision was made to remove. This is how a community driven open source project works - if nobody in the community cares about a feature it will get stale, broken, and then removed.

MR444, 2018-09-03:
@mikeller As you can see, there are some people that care. Some of them did their own tuning for making baro mode work without taking part in the community until now and many others already tried to fix it in the community. I know at least one person that tried baro mode once, noticed that it did not work (out oft the box) and flashed another firmware instead with all its other problems Betaflight does not have. It is not true that nobody cared about baro mode (and gps hold) for years. 
Examples: 
https://github.com/betaflight/betaflight/issues/3823
https://github.com/betaflight/betaflight/issues/1003
https://github.com/betaflight/betaflight/issues/6311
On RCGroups you can find old and new threads where people tried to get that mode working
https://www.rcgroups.com/forums/showthread.php?2664300-GPS-tuning-in-betaflight
https://www.rcgroups.com/forums/showthread.php?3067380-Baro-Mode-not-activating#post40025365

Variometer is also a nice feature that I miss.

mikeller, 2018-09-03:
@MR444: Ok, I stand corrected. Nobody cared enough to submit a fix that made it usable.

andrejpodzimek, 2018-09-04:
> @MR444: Ok, I stand corrected. Nobody cared enough to submit a fix that made it usable.

@mikeller That is a very unfair statement. Please let me go off-topic here and present a hypothesis on why the people who *did* care may not have come up with pull requests:

* They may not have a computer science background or a few years to spare to get it.
* Even if they do have it, they may not have the knowledge of C and JS or time to learn them.
* Even if they do know C and JS, other commitments can stop them from spending time on Betaflight.
* Even if they do have time and knowledge, their job contracts can make open-source contributions complicated.

For me the first two checklist items pass without issues, but the latter two would give me a rough time. It's definitely *not* about unwillingness to give anything back to the community. (After all, I think I've already contributed a few Mavics worth on direct donations to Betaflight and on Betaflight merchandise.)

As for the willingness to contribute code (i.e., in my case, the willingness to get over the last two checklist items), this has just bitten the dust with the recent feature removals &mdash; working against the overall aim of a project (removal of iNav-like features) is *not* a good situation for a new contributor to be in.

I definitely didn't mean to sound cynical or unfriendly here, so sorry for the tone. I just felt the need to respond to the *nobody cared enough* statement.

mikeller, 2018-09-04:
@andrejpodzimek: If you think this is unfair, then let me tell you what I see as unfair: That there are people out there who think that just because I have spare time to use, and decide to use most of it on Betaflight, this gives me (or others who are in the same situation as myself) an obligation to implement / fix / maintain the features that **they** desire.
This is not only unfair, this is also plain wrong. A lot of the work I do in Betaflight is around helping others implement the features they want - often these are features that I do not particularly desire, or have use for. But to me, 'helping' means just that - the other person leads the effort, I assist.
As for BARO, I've offered this a number of times over the past, and never had any takers. So please don't think it's unfair that nobody has come forward and offered to do this feature for the users who want it all on his own

andrejpodzimek, 2018-09-04:
@mikeller

1. Nobody used the word *obligation* here.
2. Nobody asked you to *implement* something. We merely asked that features do *not* get removed in ways that make it hard to get them back as the code around evolves quickly — especially for potential new contributors who may decide to step in in the future (e.g. iNav people, if they so choose, etc.).
3. I did *not* say it was unfair that nobody made baro improvements. I merely claimed that it had been unfair of *you* to say *nobody cared enough*. That’s a big difference in scope and meaning.
4. As documented above, this is not a matter of one user who wants a new feature all for themself. This is a matter of *multiple* users who do *not* want an existing feature (albeit a half-broken one) to be lost.
5. You say you may not want to *maintain* certain features — fair enough —, but on the other hand, a few posts above you seemed to suggest that the baro support had been somewhat unmaintained… That leaves me wondering whether it was really such a maintenance burden.
6. I deeply appreciate your free time invested into the project. I explained my position above (the last checklist item being a particularly painful one), but I’ll be more than happy to keep donating money to Betaflight, just like I did in the past, to support the project. I am not expecting free beer or whatnot.

On a more constructive note, it would be nice to have something like Kickstarter-style project bounties where people could simply offer money for features they want. That could make it easier to (1) see whether a feature is wanted and (2) find people who can work on it, for a (hopefully) fair compensation. Personally I’d be willing to chip in, say, $5k per year if interesting features were offered, just to support Betaflight.

As far as investing my free time goes — yes, I’m working on it and it may be feasible one day, but it won’t be today.

mikeller, 2018-09-07:
@andrejpodzimek:
- the obligation seems to be reasonably clear from the tone of the messages above;
- asking for broken and not working features not to be removed in favour of new and working features, or else trying to sidelines the contributors of these new features into  fixing the old and broken features before their new features get added because there is common functionality is not going to work - get over it;
- my statement was merely based on the fact that nobody cared enough to fix the broken altitude hold **for years**;
- this is not about how much work it is - this is about somebody being willing to do this work. Even if it's 5 minutes work, I can't see your attitude of entitlement helping to motivate anybody to spend these 5 minutes;
- 'donations' are generally understood as something being given without the expectation of getting something in return. If you want something done, you should consider hiring a contractor to do the work.

Closing this, as it is not productive at all.

closed 2018-09-07 by mikeller

digitalentity, 2018-09-07:
@mikeller thanks for making a nice summary :+1: