betaflight/betaflight #6678: IBUS RSSI Calculation based on package loss/corruption

IamCereal, 2018-09-01:
**Is your feature request related to a problem? Please describe.**
The TXs from FlySky/Turnigy that cannot be updated to OpenTX (e.g. **non**-FS-i6) are missing a simple way to display RSSI on OSD altough the RSSI/Error rate/SNR is calculated in the AFHDS2A protocol, such as the TGY-i10. These radios do not send RSSI on any AUX channel.

**Describe the solution you'd like**
Would it be possible to add a feature for IBUS RSSI calculation in BF based on the package loss/corruption, similar to what has been introduced for S.BUS in BF 3.4? 

**Describe alternatives you've considered**
N/A

**Additional context**
I do **not** know AFHDS2A protocol and what is actually sent, i.e. if the package loss is actually sent back to the RX or if it only calclated in the TX.

mikeller, 2018-09-02:
Not sure if this is feasible. The package loss based RSSI calculation relies on the RX sending notifications to the flight controller when frames are lost over the air, like SBus does:

https://github.com/betaflight/betaflight/blob/3a5961a7583ce59103fc565816f8c76745337dbf/src/main/rx/sbus_channels.c#L76-L85

At the moment, Betaflight is not checking for any of this for iBus:

https://github.com/betaflight/betaflight/blob/3a5961a7583ce59103fc565816f8c76745337dbf/src/main/rx/ibus.c#L177-L188

This means that this information is either not sent to the flight controller in iBus, or we are simply ignoring it. I am not an expert in iBus, and I don't have test equipment to be able to reverse engineer, so it will be up to somebody else to look into this.

atomiclama, 2018-09-03:
With stock FW on the Flysky TX this cannot be done. If you update the FW 

https://www.rcgroups.com/forums/showthread.php?2486545-FlySky-FS-i6-8-channels-firmware-patch%21

then the LQ and  I think error rate can be sent from the TX to the RX on a spare channel. I'm currently using the LQ to give me some idea of the state of the link. 
Note this is only the quality of the RX -> TX path but it is better than nothing. I've had it drop to 20% and still be fine, but it's a bit nervy.

mikeller, 2018-09-04:
@atomiclama: What we are looking into is not the 'send RSSI from the TX by sacrificing one RC channel'. It's to use the information about lost frames on the RX to generate an RX side approximation of RSSI (probably much like what is done on the TX side in the patch that you describe. If you have a look at:

https://github.com/betaflight/betaflight/blob/3a5961a7583ce59103fc565816f8c76745337dbf/src/main/rx/rx.c#L373-L379

This currently does not work for iBus, because the firmware cannot decode when a frame has been lost, and set `RX_FRAME_FAILSAFE` or `RX_FRAME_DROPPED` accordingly. But looking at the second snippet I referenced in my post above, `ibusSyncByte == 0x20` might well mean that a valid frame has been received, and other values might mean the frame has been lost.

atomiclama, 2018-09-04:
@mikeller yeah I understand what was being requested, just trying to point out that the same end result can be achieved a different way. Was not clear enough in what I was trying to get over.

I think the sync byte is just that, an iBus frame sync indicator as it is different between 2 types of RX. Maybe it changes on frame loss as you state. I'll see if I can hook a RX up and have a look.

The TX side is different as it has direct access to the registers in the RF chip so can read RSSI or calculate some form of LQ.

atomiclama, 2019-03-25:
@IamCereal Just found this: -
https://github.com/Cleric-K/FlySkyRxFirmwareRssiMod

The FW in the RX can be updated so that some form of RSSI is output on a channel. Maybe this is of use to you.

IamCereal, 2019-03-25:
@atomiclama: Awesome! Thanks for sharing! It looks really promising, I will test it as soon as I get started for real this season!

IamCereal, 2019-12-29:
Happy holidays everyone! 

After long time (being a parent of two) I have finally flashed the above solution to two X6B and one IA6B RX and it works excellent! Just want to give a huge thank you and shout out to @Cleric-K for providing this custom FW!

Best regards and happy flying!

closed 2019-12-29 by IamCereal

mikeller, 2019-12-29:
@atomiclama: Any updates on https://github.com/betaflight/betaflight/issues/6678#issuecomment-418295401?

Adding the ability to detect failsafe conditions on iBus would be valuable, as this keeps cropping up: #8877

atomiclama, 2020-01-06:
Hi @mikeller sorry no updates, lost confidence in the FS implementation by flysky and R9 reliability. So decided do design my own, Also have not been very active with the hobby recently.  Will update if I get any info.